hneemann/Digital #107: Independent sorting of device pins and truth table

roy77, 2018-01-22:
Currently the sorting of the truth table and the device pins is done by Edit=>Order....  The sorting can also be changed in the truth table Colomns=>Reorder..., but this changes are discarded after closing the dialog.

In most cases the truth table sorting is important to understand the logic or for the HEX export but sometimes I want a divergent component pinning.

The columns can also be moved with the mouse, but the change does not rearrange the rows and is therefore of little use.

hneemann, 2018-01-22:
It seems to me that there are very few use cases for such a feature.
You need a circuit that needs a specific pinning, and at the same time, you want to analyze it many times, making manual reordering of inputs a bother.

> The columns can also be moved with the mouse, but the change does not rearrange the rows and is therefore of little use.

I was not aware of this JTable feature. I will think about it.

hneemann, 2018-01-23:
A possible solution could be to put your circuit in an new subcircuit. The inner one defines the pin order required for analysis, the outer one matches the pin requirements of the embedding circuit.
Not an elegant solution, but maybe enough.

roy77, 2018-01-23:
My problem was that I use a subcircuit many times and then I find out that I have to change the pinning to get a good permanent truth table. Since I know that the setting in the truth table is only temporary, it's not a big problem for me in future. From now on, I define the pinning to get a good table and then I use the subcircuit.

I like the new mouse based sorting. It also suggests that the settings are only temporary.

hneemann, 2018-01-23:
Mouse based ordering is added in v0.17.
Persistent storage of the analysis column order is a "won't fix" for now.

closed 2018-01-23 by hneemann