hneemann/Digital #109: Documentation: Improvement of English translation

hneemann, 2018-02-04:
The English translation of the documentation could be improved.
It would be nice if a native speaker could look at it.

The source text can be found [here](https://github.com/hneemann/Digital/blob/master/src/test/resources/docu/static_en.xml).

juecoree, 2018-02-12:
I have a PR on documentation improvement for static_en.xml. I have done some improvements on the Introduction, First Step, and Wires of the static_en.xml. I hope it could be helpful to this project.

hneemann, 2018-02-14:
@juecoree Thank you for your contribution. I will integrate your suggestions soon.

aaraujo666, 2018-02-26:
Reviewed the entire text... very few modifications... but there were some...

[static_en.zip](https://github.com/hneemann/Digital/files/1757360/static_en.zip)

commit 2c311f6e (2018-02-26): improved english translation of documentation; thanks to @aaraujo666; closes #109
closed 2018-02-26 by hneemann