hneemann/Digital #115: "Set"-Input of asynchronous D-Flipflop always sets all bits?

effha, 2018-02-22:
Is this behavior correct, that the "Set" input of the asynchronous D-Flipflop always sets all bits, regardless of the value in D?
With bit length 1 all runs fine, but with longer bit lengths always all bits are set (tested with 4 and 8).
The correct value in D will only stored if clk goes to high.

hneemann, 2018-02-22:
This behavior is intended: A rising edge at the clock input stores the current state of the "D" input.
A rising edge on the "set" input, sets all stored bits to one. A clock signal is not required which makes it an asynchronous operation. If a rising edge at the "clr" input is detected all stored bits are set to zero. Also no clock signal is required, which makes this also an asynchronous operation.
What behavior did you expect?

effha, 2018-02-22:
I expected the value in D to be stored asynchronously without the clk signal. I guess I misunderstood the circle's explanation. :)

closed 2018-02-22 by hneemann

roy77, 2018-02-22:
One way to clarify this is to define the clock domains with numbers.

use C1 instead of C   (number X at end of name define a new clock domain X)
use 1D instead of D (number X at begin of name, characterize the dependency  on clock domain X)

Example:
www.ti.com/lit/ds/symlink/sn74hc164.pdf

hneemann, 2018-02-22:
This is a good idea! The naming would follow a standard.
But I suppose it would not have prevented this misunderstanding. :-)

commit f16c693f (2018-02-22): better explanation of as-ff set and clr inputs. closes #115