hneemann/Digital #117: I might be dense... but how do I make bidirectional pins work?

aaraujo666, 2018-02-24:
So... attached I have the diagram for a 28C16 EEPROM.

Dow do I make the I/O lines work bidirectionally?

The way it is right now, if I run the simulation, and set I/O7 HIGH, the wires turn red, sometimes it works (I can write the "0x80" to the EEPROM at whatever address is set on the address lines if ~CE is low and I pulse ~WE low). But that only happens, sometimes, with I/O7. Any other I/O line that I try to set, ALWAYS gives me the message.

Is there a way to make this circuit work? Is this a bug? Or am I just doing it wrong?

Is it because the splitter is "driving" the signal? Is the splitter an actual active component? I was thinking of the splitter as just a "bus" in the sense that it's an easy way to wire multiple data lines in my circuit. 

Can we have a splitter that ISN'T active? Like a "Bus", that would have the same capabilities of defining how many splits on both the input and the output, but that is really just a bundle of wires? 

Another option that I *THINK* would work, is if the current "splitter" could "propagate" Z from the inputs to the outputs. But in that case, at least for my circuit, I would have to be able to drive Z from the "driver" component.

So, is it possible to add a "checkbox" to the splitter component that "Propagate Z"? And a checkbox on the "driver" components that "Output Z when not selected" or even a "Not Selected Output Value" dropdown (where we could pick Z, 0 or 1)?

[28c16.zip](https://github.com/hneemann/Digital/files/1753721/28c16.zip)

aaraujo666, 2018-02-24:
One last comment, if you look at the values currently in the EEPROM memory, you will see the first couple of bytes have values. That shows that the circuit, occasionally, worked to write.

hneemann, 2018-02-24:
In Digital splitters have no gate delay, but they have a direction. This makes the creation of the simulation model much easier and improves the performance a lot. 
The consequence is, that they can't split a bidirectional data bus. In most cases this doesn't matter because in most cases you don't need to split a data bus.
In your special case you only need to split the bus because you want a special DIL case with only one bit per pin. 
I suppose the only thing you will do with these separate wires is to put them back together on a bus in another chip. :-)

[28c16_nee.zip](https://github.com/hneemann/Digital/files/1754284/28c16_nee.zip)

I have added an example of how this can be done with the usage of one splitter per direction. 
The OR gate ensures, that the direction of the bus is set to "output" if the EEPROM puts data to his output, which is the case if OE and CS are both high. In all other cases the two splitters behave like an "input".
Maybe i should add a special bus splitter component to simplify that. Hmm...

hneemann, 2018-02-24:
I forgot to mention: Splitters can propagate a "high Z" value. There is already a checkbox for that. 
But in this case, a splitter can have only one input, which is caused by another design decision: For a bus there is only one flag "high Z". Either all lines are in the "high z" state or none is.

hneemann, 2018-02-24:
If you use the latest [prerelease](https://infdigital.dhbw-mosbach.de/) version, also this example works:

[28c16_new.zip](https://github.com/hneemann/Digital/files/1754421/28c16_new.zip)

aaraujo666, 2018-02-24:
LOVE the bidirectional splitter!

closed 2018-02-24 by aaraujo666

LeviPesin, 2023-04-03:
@hneemann Maybe bidirectional splitters can be enhanced to support any splittings, like unidirectional?
My use case -- I try to connect 8-bit output of RAM (which is bidirectional) to 32-bit bus. I can't make this to work using current bidirectional splitters.
My current workaround is using two unidirectional splitters and two drivers, like the following:
[image]
It would be great to see bidirectional splitter with multiple bit-wires on both sides (currently only on one) and involving two output-enable signals.

LeviPesin, 2023-04-03:
> Wouldn't the illustrated application be much easier to solve with a RAM, which provides an input exclusively for writing and an output exclusively for reading?

It would remove the need for drivers, but wouldn't remove the need for splitters and will make the circuit slightly less understandable, I think... But it is a better workaround.