hneemann/Digital #118: More pins on DIL package

gercinojr, 2018-02-24:
Hello everyone!

I did the TTL 74181 in Logic and would like to put it in DIL package with 24 pins, but the program only allows at most 16 pins.

Congratulations to all for the initiative. :-)

hneemann, 2018-02-24:
There are three things to say about this:
1. You can type in any value you want. Only the combobox values are limited to 16.
2. The number matters only in very special cases. In most cases Digital determines the correct number automatically.
3. @aaraujo666 has recently implemented a [74181](https://github.com/hneemann/Digital/blob/master/src/main/dig/lib/74xx/arithmetic) :-)

gercinojr, 2018-02-24:
Thank you for the clarifications.

And regarding @ aaraujo666 having recently implemented the 74181 I already knew. That's just what made me implement it, too. I was super motivated.

And again ... thank you very much for the explanations. :-)

hneemann, 2018-02-24:
Have fun! :-)

closed 2018-02-24 by hneemann