hneemann/Digital #119: New (E)EPROM "chips"

aaraujo666, 2018-02-24:
Attached are the EPROMs and EEPROMs that I am currently using in my circuits. I put the title as 
"(E)EPROM", because the 8-Mbit version is not electrically erasable.

Just so you understand why I create all these DIL chips... I use Digital to simulate my circuits, to find design errors, before I move on to the PCB manufacturing part of my process, where I will then build these circuits with actual components.

I am in the process of building a 8-bit "CPU", using discrete components (mostly 7400 series ICs). My kids are all interested in electronics, so I am building this as a means of explaining the way things work, internally, at the processor level. 

Specs are: 
* 16 x 16-bit registers (Can each be addressed as a single 16-bit register or 2 x 8-bit registers, i.e. AX, AH & AL).
* 16-bit ALU
* 8 x 7-segment display for output
* Instruction set with 2048 op codes (Max of 16 micro-instructions per op code)
* (Under development) Video interface
* (Under development) Keyboard input
* (Under development) Serial interface

[Parallel (E)EPROMS.zip](https://github.com/hneemann/Digital/files/1754732/Parallel.E.EPROMS.zip)

hneemann, 2018-02-24:
This is a really cool project but it has a huge drawback: It's a lot of work and it's almost impossible to modify the computer if it is build. 
An alternative is to use an FPGA: You can buy a cheap development board (I use the BASYS 3), and all necessary software is available for free. With an FPGA you can play around with the processor, modify it, implement features in different ways and so on. With a few mouse clicks you can transfer the processor to the FPGA and observe it running. If something does not work, you can fix it in minutes.
If you want to do some measurements, connect that signal to a physical pin an display it on a scope. 
If you want to connect a real 74xxx chip: Design a small PCB for it and connect it to the FPGA, maybe its sufficient to use a breadboard and some wires.
Also the implementation of a video adapter is straight forward. I have recently implemented my own.

And you don't have to learn VHDL or Verilog. You can use Digital to design the computer and all the peripherals and export it to VHDL. And if your kids like it, they can learn Verilog or VHDL and build much more fancy stuff. :-)

hneemann, 2018-02-24:
I have looked to the circuits: You have not implemented the "Data Polling bit" functions.
Do you plan to use this chips in your computer without using this feature?
And why are the "not connected" pins connected to ground? :-)

aaraujo666, 2018-02-24:
These are my designs, so far...

https://easyeda.com/aaraujo666

I do have an FPGA (a Mojo V3) that I will probably use to advance my design beyond the educational realm, but for education purposes, I like the idea of separate boards, with separate functions, that can be observed while functioning, or even worked with individually (using the "Study Board" in my project). 

The one issue I have with the Mojo is that the toolset it uses (Mojo IDE) only uses Verilog and Lucid. Granted that it uses Xilinx in the background to do the synthesis, etc., but the Mojo IDE is much more user friendly :)

As for the circuits I uploaded, I don't intend to use Data Polling, but is that something that could be implemented in Digital?

hneemann, 2018-02-25:
I do not know Mojo, but Xilinx has released a new design suite called Vivado, which is much more user-friendly than ISE. So it is possible to use Digital as an IDE to build the circuit, and to use Vivado as a tool only for the synthesis and programming of the FPGA. The user does not even see the VHDL code. But Vivado only supports the newer FPGAs and Mojo 3 uses a Spartan 6, which Vivado does not yet support.

If I get to it, I will look at the data sheets of the 28c512 and try to implement the data polling. But if you don't plan to write to theses chips from your computer, data polling doesn't matter.

hneemann, 2018-02-25:
I have looked at your schematics: You are planning to build a microcoded multi cycle computer?
Why not build a simpler single cycle risc machine?

commit 836f21bb (2018-02-28): added global ROM settings. closes #119, fixes #121
closed 2018-02-28 by hneemann