hneemann/Digital #121: Embedding

aaraujo666, 2018-02-26:


hneemann, 2018-02-26:
That's an interesting problem. I have to think about it.

hneemann, 2018-02-27:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) build.

aaraujo666, 2018-02-27:
Will do

aaraujo666, 2018-02-27:
Not sure what you want me to check? The latest version still does not allow
me to edit the ROM contents unless I "Open Circuit" and edit it there. And
if I do that, all instances of that "chip" now have that contents.

hneemann, 2018-02-27:
In the circuit attributes settings you now can define different contents for the used ROMs depending on how many times a ROM was included.
But I have not pushed this feature branch yet. So you have to use the latest [pre-relese](https://infdigital.dhbw-mosbach.de/) build.

aaraujo666, 2018-02-27:
Ok. Will check. Am using pre-release version.

aaraujo666, 2018-02-27:
Can't get it to work... Edit/Edit Circuit Attributes/Edit gives me an empty
"Used ROM's" window.

I'm using EEPROMs? Is that why?

hneemann, 2018-02-27:
I still need a unique identifier for each ROM.
Therefore, the label of the ROM must be set to '*', which will be replaced by a corresponding identifier.
All circuits must also have different labels to create the unique id. The ROMs / EEPROMs appear in that window. 

[image]

I suppose I have to rework the help text. :-)

aaraujo666, 2018-02-27:
I’ll try again. But I think I tried with different labels. And even with *
as the label

aaraujo666, 2018-02-27:
[image: Inline image 1]

[image: Inline image 2]

[image: Inline image 3]

hneemann, 2018-02-27:
Can't see the images but try this test case: [rom.zip](https://github.com/hneemann/Digital/files/1764198/rom.zip)
Does it pass?

aaraujo666, 2018-02-27:
yes...

But again, these are ROMs... my 28Cxx circuits use EPROMs. Is that the
reason it doesn't work?

aaraujo666, 2018-02-27:
try the TEST.dig I attached to the last email

hneemann, 2018-02-27:
In the example I just sent, I replaced the ROM with an EEPROM, and that also works fine.
The email attachment did not make it to GitHub.

aaraujo666, 2018-02-27:
Ok... figured it out...

The actual EEPROM component *INSIDE *the 28c16.dig file needs to have a
label of "*" or something unique...

aaraujo666, 2018-02-27:
It would be nice to be able to load .hex files from the "Used ROMs" Edit
screen. As is, you can only input the ROM contents manually...

hneemann, 2018-02-27:
Try the "Data" menu in the edit dialog.

aaraujo666, 2018-02-27:
ok... that works... didn't notice it... it doesn't look like a button or a
menu item :)

hneemann, 2018-02-27:
In the latest pre-release the menu is named "File" which is more intuitive. :-)

aaraujo666, 2018-02-27:
Helmut,

Can you contact me at anthonyaraujo@gmail.com? I have some questions/suggestions that might best be discussed in a non-public space...

Thanks,

Tony

hneemann, 2018-02-28:
You can send a private message to digital-simulator@web.de.

closed 2018-02-28 by hneemann

xotmatrix, 2020-06-20:
These processes do not appear to be documented.

(1) creating a ROM component, labeling its embedded ROM device with "*", locking it, saving it 

(2) inserting the new component into a circuit, labeling it uniquely, loading data into it from "Circuit specific settings"

commit 836f21bb (2018-02-28): added global ROM settings. closes #119, fixes #121