hneemann/Digital #124: Presettable counter

aaraujo666, 2018-03-18:
I implemented a presettable counter (attached).

It would be nice if there were a presettable counter component along side the existing counter component.

Mine is very specific to my implementation, but if you could implement a presettable version, that would be awesome.
[Presettable Counter.zip](https://github.com/hneemann/Digital/files/1823128/Presettable.Counter.zip)

hneemann, 2018-03-19:
You mentioned the problem in your feature request already:
There are so many options dependent on the specific use case. It's difficult to add a built-in counter every body is happy with. And at the other hand: it is very easy to build a counter with a view components that fits your specific needs.

Which inputs/options are at least necessary to get a more usable "universal" counter:

- a clock input
- a enable input
- a clear input, synchronous or asynchronous
- a load input
- the data input to load a value
- a direction input
- min and max values? Or are these attributes of the counter?
- if min/max is reached: stop, reset to max/min, continue counting, toggle direction?

For inspiration one can look at an AVR uC: There are many pages of documentation to explain all the different modes of operation of the timers/counters.

hneemann, 2018-03-19:
After thinking about it, I decided to add an improved counter that could possibly meet 90% of user needs..

closed 2018-03-19 by hneemann