hneemann/Digital #127: Support for polarity aware LEDs?

n3rdopolis, 2018-03-29:
Hi

One feature that it might be worth simulating is to simulate the polarity of the LED, so that you can have  gates controlling both poles, like an example shown below. I guess the closest to this is the lightbulb, but those don't seem to be dependent on polarity, like LEDs are
[image]

hneemann, 2018-03-29:
Do you have any idea what the graphical representation should look like?
I think the diodes symbol should be visible, but also the state "lights" must be clearly visible.
The above figure does not show the condition "lights" very clearly.

n3rdopolis, 2018-03-29:
maybe the inside of the triangle could change color
or maybe better, ones that care about polarity can be a semi circle like this 
[image]
since usually the base of a real LED, the cathode side is marked by the flat side of the base

hneemann, 2018-03-29:
Can you explain what your use case is? What kind of exercise do you want to do?

hneemann, 2018-03-30:
@n3rdopolis 
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/). What do you think about it?

n3rdopolis, 2018-03-30:
Awesome!

hneemann, 2018-03-31:
added in v0.18

closed 2018-03-31 by hneemann