hneemann/Digital #128: Small GUI improvements for Eduaction

TimFFischer, 2018-04-04:
I'm starting to use the program in the beginners course of digital electronics, for with digital is much better, than most of the other freeware solutions. 
However, I miss some convinience in the user interface:

- starting a simulation, one can change an multi-bit input via double-click -> change of the value -> pushing return (all without using the mouse). However, this is only possible once. If the value was already changed before, pushing return does not change the value.
- similar situation in the input form: a possibility to change the value without the use of the mouse and without closing the form would be great (e.g. via Ctrl-Return or Shift-return).
- an general option to force the number prefix (e.g. 0x10, 010, 0b10) for the shown values would be perfect

hneemann, 2018-04-04:
> starting a simulation, one can change an multi-bit input via double-click -> change of the value -> pushing return (all without using the mouse). However, this is only possible once. If the value was already changed before, pushing return does not change the value.

Can you explain more exactly what you did? I have tested this on Win7 and Debian and it works as expected.

> similar situation in the input form: a possibility to change the value without the use of the mouse and without closing the form would be great (e.g. via Ctrl-Return or Shift-return).

Same here: You can press return to close the attribute dialog. This is also working on Win7 and Debian.

> an general option to force the number prefix (e.g. 0x10, 010, 0b10) for the shown values would be perfect

You can choose the format of the values shown at the inputs and outputs. Or do you mean the format of the values in the input dialog?

TimFFischer, 2018-04-04:
Thank you for your fast answer. I just realized, that the first improvement is already available in the version v0.18. I used the version v0.16. Sorry.

The second improvement would be beneficial in a situation, explaining the results of a digital circuit. There, I would like to change the input attribute (e.g. from 0x01 to 0x02 and 0x08 and 0xFF) by writing the number and confirming with a keystroke (e.g. <Ctrl>+<Return>)  without closing the attribute dialog and without the mouse. So, similar to writing the number and pressing <Return>, but with out exiting the dialog.

Clarification for the last issue: for the input dialog I understood, that it is possible to switch between the format via the prefix 0x or 0. I recommend to my students to use the format prefix at any time in order to avoid misinterpretation. However, this is not alwas the case in digital.

Simplified example: 
- 8-bit input with number format "hex" connected to 8-bit output with number format "default". 
- Result in simulation for input 0x10: "10" on input, "0x10" on output. Input "10" could be misinterpreted as decimal.
-> requested result: all outputs shall be printed with present number format prefix, when a general option (e.g. in settings) is activated.

hneemann, 2018-04-04:
> Thank you for your fast answer. I just realized, that the first improvement is already available in the version v0.18. I used the version v0.16. Sorry.

Never mind: There was a bug in v0.16 that was already fixed in v0.17. :-)

> The second improvement would be beneficial in a situation, explaining the results of a digital circuit. There, I would like to change the input attribute (e.g. from 0x01 to 0x02 and 0x08 and 0xFF) by writing the number and confirming with a keystroke (e.g. +) without closing the attribute dialog and without the mouse. So, similar to writing the number and pressing , but with out exiting the dialog.

I have added a shift+enter hot key to the single value input dialog. Try the latest [pre release](https://infdigital.dhbw-mosbach.de/) version.

> Clarification for the last issue: for the input dialog I understood, that it is possible to switch between the format via the prefix 0x or 0. I recommend to my students to use the format prefix at any time in order to avoid misinterpretation. However, this is not alwas the case in digital.

The idea is that if the user has explicitly selected a particular format, it is not necessary to flag that format with a prefix. In this case I can save the space on the screen.
But perhaps this is not a good approach. I have to think about it.

commit d48dd576 (2018-04-05): Bin and hex prefix is always shown. closes #128
closed 2018-04-05 by hneemann