hneemann/Digital #129: Suggestion: Automatic gate step at time interval

michaelmortensen-m4y, 2018-04-05:
I like using Digital (v0.18) to simulate and visualize asynchronous sequential circuits, and it does so much better than other similar apps and it looks very cool. Asynchronous circuits use handshaking between their components in order to perform the necessary synchronization, communication, and sequencing of operations so the normal start button gives an error because it checks for oscillation. This means that I can only simulate f.ex. a Muller pipeline using the single gate step mode and then continuously click the calculate single gate step button to see what is going on, so i have some ideas for future improvements:

Right now i am using a AutoHotKey script to click the calculate single gate step button once every 50ms or some other time interval so it would be cool to have a simpler way of simulating in gate steps at some time interval and be able to start, stop, pause, change step rate etc. and still be able to manipulate inputs between steps.

It would also be cool to not have a maximum delay of 20 steps for the delay elements so that we don't have to make embedded circuits of long series of delay elements when matching handshaking signals with the delay of combinational logic between latches.

I have attached a fun little model of the Muller pipeline :) 
[AsyncTest.zip](https://github.com/hneemann/Digital/files/1880426/AsyncTest.zip)

hneemann, 2018-04-05:
> I like using Digital (v0.18) to simulate and visualize asynchronous sequential circuits, and it does so much better than other similar apps and it looks very cool.

From the simulation model point of view, it is nothing special to simulate a circuit that never goes into a stable state! I simply did not have the idea that someone could use a circuit in which this is not an error. Although it is obvious.

> Right now i am using a AutoHotKey script to click the calculate single gate step button once every 50ms or some other time interval so it would be cool to have a simpler way of simulating in gate [...]

I'll think about a more elegant solution than the one with an AutoHotKey script. :-)

> It would also be cool to not have a maximum delay of 20 steps for the delay elements so that we don't have to make embedded circuits of long series of delay elements when matching handshaking signals with the delay of combinational logic between latches.

What values do you think about? In the moment there is a upper limit only because simulation of large values is expensive. And your example uses a maximum delay value of 4 steps.

michaelmortensen-m4y, 2018-04-05:
> I'll think about a more elegant solution than the one with an AutoHotKey script. :-)

That would be awesome ;) 

> What values do you think about? In the moment there is a upper limit only because simulation of large values is expensive. And your example uses a maximum delay value of 4 steps.

The example was just to show the pipeline with no combinational logic between the latches. I didn't realize that a single delay element with a long delay would have a larger impact on simulation performance than a series of multiple delay elements having the same equivalent delay. If so, then the latter way is clearly preferred.

hneemann, 2018-04-06:
@michaelmortensen-m4y 
Please try the [pre release](https://infdigital.dhbw-mosbach.de/) v0.18-6. It has a new component to define the behavior if a asynchronous sequential circuit is simulated (components->Misc->Async).
Please let me know what you think about it.

michaelmortensen-m4y, 2018-04-07:
Nice! This was exactly what i was thinking about and i am impressed that the simulation can run at such high step rates. This allows for cool animations and it also works with embedded circuits so that is perfect.

For clocks to work automatically in gate step mode i am using small logic oscillators with delay elements to control number of steps ON/OFF. This works very well but it would be nice to have a simpler way of defining a clock signal for step mode. 

Digital is now a great tool to easily simulate and visualize modern network-on-chip with several different clock domains and asynchronous clock domain crossing and other cool modern stuff that includes both synchronous and asynchronous techniques :)

closed 2018-04-07 by hneemann

hneemann, 2018-04-07:
For now I do not want to implement a special async clock component. And i also do not want to extend the available clock component to make it usable in async mode.
For now the user has to build his own oscillator which gives full control about phase, on-time and off-time.

I've added a Muller-pipeline example. The used Muller-C-element is a bit simpler than your approach. Maybe it's interesting for you. :-)

commit 2d718dca (2018-04-07): updated release notes, closes #129