hneemann/Digital #130: Group box

aaraujo666, 2018-04-06:
Would it be possible to have a "group box" ?

What I mean by this is just a visual box that could be used in the circuit to separate/designate areas of the circuit.

Ideally, anything within the box would be bound to the box so if I move the box, all the components in it move with it, but that's not 100% necessary. We can always do a rectangular select that includes the box and it's contents and move it that way.

If possible, it would also be nice if the box has it's own "Label" attribute which would display right above the box. Ideally, position of the title could be an attribute as well (i.e. "top", "bottom", "left", "right")

Thanks!

hneemann, 2018-04-07:
It takes a lot of work to add arbitrary graphical elements to the component area.
That's why there are no custom shapes yet. (see #39)
You must allow the user to interactively change the shapes with the mouse. Something that does not exist yet. In addition, it makes little sense to add only rectangles. It should also circles, ellipse, etc. be possible.

hneemann, 2018-04-07:
see #131

closed 2018-04-07 by hneemann