hneemann/Digital #133: Export to schematic or netlist

PiMaker, 2018-04-09:
I have created a CPU in Digital that I now want to make a reality in 74xx logic. For that I would like to have a circuit diagram readable by a program such as KiCAD or CircuitMaker (I want to design some PCBs as well).

I think it would be beneficial to allow export to such a format, for example the KiCAD standard .sch defined [here](http://kicad-pcb.org/help/file-formats/).

Now, I understand this might be a rather big request, so first I looked into writing a converter program myself (basically reading in .dig files and outputting .sch). The issue I faced with that is that your current format (.dig, XML) is not very easily parseable for such requests, as only visual positions are encoded (as far as I understand it, anyway). Is there a way to export something like a netlist with connected components, or any other, more abstract and machine-readable representation of a circuit?

Thanks in advance!

hneemann, 2018-04-10:
Thats a bit tricky:
The creation of a schematic is easy because all the positions of the components and the wires are available in the dig file. The problem here is that no EDA suite offers components at a DIL level. If you are using a 7400 than there are 4 separate NAND gates to draw and to wire, not a single 7400 DIL shape.

On the other hand: The creation of a netlist is implemented in Digital because a netlist is necessary to run the simulation. But the netlist used by Digital goes all the way down to the gate level. So using a 7400 or a more complex chip like a 74181 means the netlist contains dozens of gates, not a single 74181.
Thus creating a schematic seems very difficult to me. The creation of a netlist could be implemented using a modified algorithm which stops going further if a embedded circuit represented by a DIL shape is reached.

And of course there is a more abstract and machine readable representation of a circuit available. It is called VHDL and you can run it on a FPGA. :-)

hneemann, 2018-04-16:
I decided not to implement such a feature.
Most courses where 74xx circuits are built will do this on a breadboard. Very few will produce printed circuit boards. And who wants to realize more complex circuits, uses an FPGA.
I'm afraid such a feature is not worth the effort.

closed 2018-04-16 by hneemann

mixotricha, 2022-04-11:
I know this is an old thread. But today I had a case that was the reverse of this where I would like to bring in an existing schematic from KiCAD. I had the idea that rather than trying to figure out the wires if it just created a tunnel between every pin that would be a start. Also if a mapping for some component did not exist it would just leave it out. I wondered if it might be easier to go from foot prints in a KiCAD PCB file to a DIG rather than from a schematic.

hneemann, 2022-04-11:
@mixotricha 
To transfer an existing circuit from KiCad to a dig file seems to me to be an even rarer special case than the reverse.