hneemann/Digital #134: Did splitter functionality change?

aaraujo666, 2018-04-10:
I had a circuit that was working... and then it stopped.

There are many splitters in this circuit, all of which had pull-down resistors on their inputs. I though they were required on the input of all splitters (I vaguely remember an error about that previously, so I always put pull-downs on the inputs of splitters)

I removed all the pull downs, and now the circuit works again...

Did the functionality change? Or am I using the bidirectional splitter when I think I'm using the regular splitter? Is there a way to see which type of component is in the circuit? Or do I have to look at the XML?

Thanks!

Tony

aaraujo666, 2018-04-11:
Did not notice that in the release notes... sorry bout that...

As for the circuit, I don't have the version that doesn't work. I'll see if I can work up an example that demonstrates the problem...

Thanks,

Tony

hneemann, 2018-04-25:
For now I will close this ticket.
But do not hesitate to reopen it when there are difficulties again.

closed 2018-04-25 by hneemann