hneemann/Digital #137: Digital.sh can't find Digital.jar File

Monotoba, 2018-04-19:
On Ubuntu calling Digital.sh errors with failed to find Digital.jar. Indeed there is no Digital.jar file in the repo clone. Does the Digital.exe need renamed?

Monotoba, 2018-04-19:
> You're right, the repo does not contain a Digital.jar file. The repo
> contains only source code (except the exe wrapper).
> You can either download the distribution zip file from the latest release
> <https://github.com/hneemann/Digital/releases/latest> or create the jar
> file by calling mvn install from the command line in the folder that
> contains the pom.xml.

Thank you for the info. I'll create the jar file.

hneemann, 2018-04-20:
If you are interested in trying the latest changes, you can also download the latest [pre release](https://infdigital.dhbw-mosbach.de/) build.

hneemann, 2018-04-20:
Have fun! :smiley:

closed 2018-04-20 by hneemann