hneemann/Digital #138: Terminal and Keyboard Component I/O

spdavies, 2018-04-19:
The Terminal component would be easier to use if it had a write enable input. In some courses students are discouraged from using gated clocks. A gated clock is difficult to avoid with the terminal component without a write enable input.

The Keyboard component is very difficult to use effectively without data available output. There is no way to know if the keyboard has character(s) without selecting it. A read enable input would also be helpful.

hneemann, 2018-04-19:
Why is the keyboard hard to use?
You can set the enable input, and if the data output is zero, no character is available.
When the enable input is low, the output is in high Z state. 
You can easily connect the keyboard to the data bus of a processor.

spdavies, 2018-04-19:
That is true as long as the design allows for polling the keyboard.

hneemann, 2018-04-19:
That's a good point! :-)
An interrupt-driven keyboard is difficult to implement. I will think about it.

spdavies, 2018-04-20:
For the class assignment I was working on, simply adding a read enable input to the Keyboard component is sufficient. That way students don't need to use gated clocks. They are polling the keyboard anyway so checking for the output being non-zero is not a problem.

hneemann, 2018-04-20:
From my point of view, the keyboard does not need a clock. It works similar to a ROM, which also does not need a clock. And what function should a read enable input have compared to the already existing select input?
When the keyboard is connected to the data bus of a processor, it is not necessary to connect the clock to the keyboard. Not even by some gates. So where is the gated clock?

I do not get it. Maybe you should explain your use case in more detail.

hneemann, 2018-04-21:
You are right! 
The keyboard component as it is now is not able to deal with hazards at the select input!
I am surprised that I did not immediately realize that!