hneemann/Digital #145: Remove unused state bits before creating jedec files

roy77, 2018-05-08:
In order to explain automats, I carry out an experiment on a simple Medvedev automat. At the beginning only some states are defined, later more. I don't want to change the hardware, the students just have to change the LUT with the transitions of the states. I have attached an example. I cannot create a "jedec" file until I have described enough states, otherwise I get the error "de.neemann.digital.builder.jedec.FuseMapFillerException: It is not possible to set physical pins to constant values!" 
Is it possible to remove rows in the truth table with static values before generating the jedec file?
[MedwedewTest.zip](https://github.com/hneemann/Digital/files/1985010/MedwedewTest.zip)

hneemann, 2018-05-08:
One should better find out how to create constant pins in TT2.
That would solve the root cause of the problem.

roy77, 2018-05-09:
The fitter accept an output which is independent from input values and as a consequence the output is constant. With phase the value of the output can be changed.

.i 3
.o 2
.type f
.ilb A B C
.ob Y Yconst
.phase 10
.p 3
000 1-
111 1-
--- -1
.e

I have no board here, but the fitter accept the example.

hneemann, 2018-05-09:
I have created a CUPL circuit with constant outputs. This is what WinCUPL creates: 

```
.i 2
.o 4
.type f
.ilb  B6 B7 
.ob   A4 A5 A6 A7 
.phase 1111
.p 2
11 1000
-- 0101
.e
```

In this example is A5=A7=1 and A6=0.
Its not a big deal to implement that behavior.

hneemann, 2018-05-09:
@roy77 
Feel free to play around with this implementation.
But be careful: It's not yet tested on real hardware.

roy77, 2018-05-09:
My first impression is good. I will test it extensively with hardware on Friday.

roy77, 2018-05-11:
My samples are running now. Thanks.
I have found a problem.
In the attached example there is a switched , a constant 0 and a constant 1 output. 
[AtmelImplementation.zip](https://github.com/hneemann/Digital/files/1994655/AtmelImplementation.zip)

The logic in the tt2 file is identic for constant 1 and constant 0
.i 1
.o 3
.type f
.ilb SW1
.ob A1 LED1 LED2
.phase 111
.p 2
1 100
- 000
.e

For constant 1 on LED1 the logic should be
.phase 111
.p 2
1 110
- 010

or

.phase 101
.p 2
1 100
- 000

hneemann, 2018-05-11:
@roy77 
Now it should work correctly.

roy77, 2018-05-11:
Yes, it works.

closed 2018-05-11 by roy77