hneemann/Digital #150: 74138 library part bug

dfbg, 2018-05-15:
When driving this part with a counter and hooking up the higher bits to the enable pins of the 74138 it will leak from output Y0 just before the disable takes effect.

I've attached a demo circuit along with a logic level 74138 circuit modeled from the TI datasheet.  You can watch the Y0 toggle flip-flops behave differently between the two parts.

[demo.zip](https://github.com/hneemann/Digital/files/2003351/demo.zip)

hneemann, 2018-05-15:
That's an interesting timing issue. 
Do you know how real ICs behave?

dfbg, 2018-05-15:
I don't yet.  I just got a bunch of parts to start breadboarding.  I'm not sure I'll be able to wire something up before I get busy for a week.  I'll let you know when I get a chance.

hneemann, 2018-05-15:
Do not do the many work of trying it out.
I'm pretty sure the current library part behaves wrong.

dfbg, 2018-05-15:
I plan on breadboarding most of this project so the test would be a small side test along the way.  But I won't prioritize it.

It also looks like the 74238 part has the same problem.

hneemann, 2018-05-15:
Yes, it has! Most current 74xx Plexers show this behavior.
I'll take a look at all the Plexers again.

commit c23895b2 (2018-05-15): fixed timing issues in the 74xx plexers, closes #150
closed 2018-05-15 by hneemann