hneemann/Digital #151: Changing the name

vladtyum, 2018-05-17:
I'm sure it sounds a bit too extreme, but please give it a short consideration. 
As a guy who really loves your SW, I'd like to put my files on web, and be able to search and download them online. Yet googling "digital" gives us huge noise, everything but not what we need. 
I'd like to suggest slightly change the name, probably add couple letters after the name something like  DigitalHW,  DigitalSM, DigitalHWSM DIgital[whatever] just to make it easier for us to be able to search it as "Logism" and build community of enthusiasts around it. 
Much love.
Vlad

xotmatrix, 2018-05-17:
I've resisted posting an issue like this but I agree 100%. "Digital" is unsearchable. I once lost track of this delightful software and it took me ages to find it again.

sgparry, 2018-07-21:
HNDigital in honour of Helmut? He deserves maximum credit in the name.

LarsiParsii, 2018-08-15:
A great simulator, but I agree. I wouldn't have found this simulator if it hadn't been for one of my teachers, as it's impossible to search up on the web. :/

kal9001, 2018-08-23:
Yes. I have literally just had this problem.
I found this software just today while trying to figure out how to get logisim-evolution working with GHDL, and someone suggested this project. but after closing a tab it was next to impossible to re-find it without digging through my history.
Googling "digital" quite rightly gives extremely broad results, even trying to qualify it with "simulator" and other such things doesn't help at all.

nevercast, 2018-08-24:
Same issue here, me and a friend of mine have built many the CPU in Logisim and when we discovered this software we were overjoyed, but when Uni came around we lost it. GoogleFu "Digital Java Logisim but better" didn't help. Glad I've got it Starred now though!

fdedinec, 2018-11-15:
In my experience, google Digital Logisim, which works but is too much advertising for logisim.

vladtyum, 2018-11-17:
@hneemann what's your opinion on that, mister ?
Many thanks for your commitment and making the project going 🖖

hneemann, 2018-11-19:
@vladtyum 
I'm thinking about "DigitalCS" which stands for Digital Circuit Simulator.

augu5te, 2018-11-23:
+1, I redicovered Digital (I've forgotten it, not Logisim).
IMHO, having "Sim" in the name is more explicit, CS equal implicitly Computer Science.
Great project in any case !

shawty, 2019-02-11:
As a longtime Logisim user, and now "Digital" user, I concur.  I find myself having to use "github digital" or "hnee digital" (I can never remember full name) just to find it when looking for the github page.

Since your essentially "Logisim" the next generation, you could maybe also pay homage to Logisim somwhere along the line too.

hrobeers, 2019-11-13:
Another important reason to change the name is that it is impossible to search for component libraries online.

e.g. I'm looking for an SPI serial EEPROM, but there is no obvious search term to figure out whether someone might be sharing one online.

harwoodr, 2020-01-27:
The Esperanto word for "digitial" is "cifereca" - relatively unique on the internet.

hneemann, 2020-01-28:
@harwoodr 
My current favorite is "DigitaSi" because of the remark from @augu5te.

JarrettBillingsley, 2020-03-07:
I was gonna suggest "Digisim" to make it an homage to "Logisim", but it seems there's already some software named that :(

fuomag9, 2020-04-02:
I agree with the issue, if "digital github" would not have yielded results I'd never have found it

charlespax, 2020-07-26:
I came here to file this exact issue. Digital is pretty easy to use, so it feels like a simple simulator. What about calling it "simsim?" Google returns fewer than fifty results.

[image]

jwbruce, 2020-10-05:
I will have to pile on here too.  Digital is a great little tool.  My students learn a lot about combinational and sequential design by using it in my class.  However, the name "Digital" makes the tool very hard to locate on the internet, find any other references to tutorials or examples by 3rd parties, AND when discussing Digital in my digital systems class about digital design and digital logic gates it can get very confusing.  If there is a desire to not to change the name too much.... how about "Digitalis"?  Really anything other than a word that literally is an adjective modifier found in 50% of online subject matter.

jay20162016, 2021-02-08:
This is unoriginal, but what about "Logisim Digital" as a name?

TheWildJarvi, 2021-04-29:
Okay, I have heard numerous complaints about the name being unsearchable on google. My suggestion, "Digisim".

sgparry, 2021-05-05:
Is three years enough for time this discussion?
@jwbruce Digitalis certainly gets the heat pounding...
@hneemann did you deliberately leave out the l from DigitaSi?

peterb12, 2021-05-17:
FYI I've been doing a YouTube series using Digital and "Where can I find this?" is the number one question people have. They absolutely are unable to locate the software by typing "digital" into google. A name change to something searchable would really help this software's adoption, IMO.

charles-fox, 2021-06-29:
I came here from logic-evolution's page, which is still the only way I can find it!    Currently looking at both and comparing, logic-evolution seems to have the bigger community around it, which is important, in part due to this naming issue?

trayres, 2021-08-04:
This project is amazing and deserves a name that is more discoverable.

mspeculatrix, 2022-01-09:
Just want to add my +1 (the first of 2022) for a name change.

Rishav-mngo, 2022-02-05:
The name is really hindering the popularity. Please consider changing the name.  +1

NotCoolKiwi, 2022-05-11:
Amazing piece of software but can't find anything on google. Possibly consider changing the name. Another solution could be to create some kind of database or built-in tool where people can upload their own chips and such. This way, we may be able to find what we are looking for easier. As the naming may also be hindering the popularity, maybe consider doing both. Thank you so much!

sgparry, 2022-08-01:
I think we need to up-vote on specific names, rather than just "we want a new name" so: HNDigital + 1...

shawty, 2022-08-01:
+1 for HNDigital - works for me.

CMDR-Piboy314, 2022-08-02:
+1 for HNDigital

rejahtavi, 2022-09-12:
HNDigital looks good to me, but if you're against including your initials, or want something more creative, I suggest "Aristottl".

It's a mashup of TTL and Aristotle, the Greek philosopher known for his work on early formal logic. As a bonus, Google finds even less for it than HNDigital.

sgparry, 2022-10-11:
I still think HNDigital works well as a name; Another possiblity: HEDSim; Highly Efficient Digital Simulator.

ekm456, 2022-11-02:
Please change the name to make it searchable.

charlespax, 2022-12-21:
This has been an issue sine 2018. At what point should someone fork Digital, rename it, and consider Digital as the upstream of whatever is the new name?

gresolio, 2023-06-18:
Google stubbornly refused to find this project, giving many irrelevant links...
A successful search query for me is "logisim alternative digital", which finally led me to the post with the correct link:
https://www.reddit.com/r/logisim/comments/kv9qkg/digital_a_logisim_alternative/

chuckhoffmann, 2023-06-22:
The search I use is "Neemann Digital". Seems to work pretty well, and I just have to remember Herr Professor Neemann's last name. Maybe just formalize the name of the software as "Neeman Digital"?

shawty, 2023-06-22:
@chuckhoffmann yea that's what I use too.

salewski, 2023-08-07:
+1 for HNDigital

callegar, 2023-11-01:
As an advice for the time being, the practical way to tell others how to search for this software is to tell them to google for "digital+git" (comes out as 1st result). In general, when you look for software and you know it is opensource, `name+git` is a nice way to filter out a lot of noise.

jxu, 2023-12-05:
HNDigital or Digisim - really anything would be an improvement
@hneemann

dragostego, 2023-12-05:
It's truly impossible to search. even easy things become impossible meaning every question requires checking the documentation.

+1 arristottl or HNDigital

9EED, 2024-07-25:
a bump to this, it took me ages to find this repo after someone told me about this project

jinnturtle, 2025-02-10:
I wonder if the project was intentionally meant to be less discoverable, if not - I vote for an alternative name also.

shawty, 2025-10-15:
for a long time now I've been using "neeman digital", works pretty reliably for me, most times coming top in my search

[image]

Indecently, yes that video in the results does also make a point of being hard to find too.

https://www.youtube.com/watch?v=o7TQ-c1WNHg

EzraWolf, 2025-12-11:
+1 for Neeman Digital/NMDigital/NMD. I have been trying to find this software for *entire months* again.