hneemann/Digital #152: Components->Custom->Update can become unreachable

dfbg, 2018-05-24:
If there are many top level custom circuits the menu can get too tall to fit (especially when the circuit images are added) and the Update option can end up off the bottom.  Since I can't get the menu to scroll I can't reach that menu item and I can't get circuit updates without restarting to shrink the menu.

I don't know how to solve the scrolling problem but it could be mitigated by moving the Update item to the top of the menu.

hneemann, 2018-05-24:
Please try [v0.18-66](https://infdigital.dhbw-mosbach.de/).
What do you think?

hneemann, 2018-05-24:
Or do you prefer [v0.18-67](https://infdigital.dhbw-mosbach.de/)?

dfbg, 2018-05-24:
66 for sure.  Both because it matches the IO menu and because *everything* isn't another level down.

It does make the 74xx->Basic collection strange in the Component Tree because it breaks them into 3 levels.  Maybe build separate organizational constructs for each to take advantage of their differences.  Even when the menu is too tall I can still get to the items via the tree in 0.18release.

Is there a way to remove the icons from the menu items?  Maybe only show the last 3 or 5 selected parts in a menu so it has less of a growth factor.  Then there could be more items before a sub-menu is needed.

hneemann, 2018-05-24:
I think i will use a mix of both: Not everything goes a level down, by keeping the first entries at the top level and the menus don't become too deep by groping the other entries like in 67.
You can try it in [v0.18-68](https://infdigital.dhbw-mosbach.de/).

dfbg, 2018-05-24:
Can you make the cut-off dependent on the screen size?  Even with all the icons in place there's still room for at least two more entries on my 1080 screen.

I think filling the top level menu works better than balancing the count in the main and sub-menu.  My current Custom menu has 2 folder sub-menus, then 7 circuits, then a sub-menu with 7 circuits.  The 74xx->Basic has 11, 11, 9.  I'd rather have as many at the top level as possible for quicker navigation.

The algorithm also seems to not count folder sub-menus in the split count.  So it would be possible to grow the menu too tall with lots of file organization.  Possibly also true if there's so many circuits that there are too many group menus.  But those might be too extreme to worry about at this point.

hneemann, 2018-05-31:
Making the menu size depending on the screen size adds a lot of ugly code. I do not want to do that.
Filling the top level menu and balancing only the sub menus is not a big deal. You will find that in v0.18-69.

Taking also the sub folders into account to create a perfect balanced menu adds a lot of complexity and adds little value. The user can easy reorganize his circuits.

I expect v0.18-69 becoming the final solution.

commit ac063e2b (2018-05-31): limits the menu size in the component menu. Closes #152
closed 2018-05-31 by hneemann