hneemann/Digital #154: Simulation Measurement graph should show Z state

dfbg, 2018-05-27:
It's currently hard to confirm the behavior of a circuit with open collector outputs where the pull-ups would be outside the circuit.

dfbg, 2018-05-28:
This is when I'm working on building a circuit and testing it by hand along the way.  It's all very dynamic and a test file doesn't make sense yet.

Much like a running circuit uses dark green, light green and grey for the wire states I'd like to see the Z state in the graph so I don't have to watch the wires as they change.

hneemann, 2018-05-28:
What is your idea of how to represent the hi-z state in the measurement graph?

dfbg, 2018-05-28:
I think drawing Z values in grey (maybe the same as the circuit wires, maybe a little lighter for better contrast) at the last non-Z level.  That way it doesn't imply a value change as well.

Or, because different hardware technologies (TTL, CMOS, etc) behave differently with floating wires, just don't draw the line.  And maybe draw a very light "Z" instead so it's clear it's not broken.

commit eb5cf1e2 (2018-05-31): Measurement graph omits the line if in high-z state. Closes #154
closed 2018-05-31 by hneemann