hneemann/Digital #156: Can not analyze a cmos circuit with current version (v0.16 works)

roy77, 2018-05-31:
With the current version (Build git-Revision: v0.18-68-gaa207fb) I cannot analyze the following circuit.
[A2.zip](https://github.com/hneemann/Digital/files/2059499/A2.zip)

Message:
de.neemann.digital.analyse.CycleDetector$CycleException: The circuit contains cycles. It's not possible to analyze such a circuit.
        Cycles arise if an output of a gate is fed back to one of the inputs of the same gate.
        The use of switches, FETs or relays also causes cycles.
        at de.neemann.digital.analyse.CycleDetector.checkGraphForCycles(CycleDetector.java:134)

Correct analyze is possible with older versions. Tested with Build git-Revision: v0.16

hneemann, 2018-05-31:
That's not a bug! That's a feature. :-)
I decided to detect cycles in circuits and to deny the analysis of such circuits to avoid wrong results. 
If you build a flip-flop from e.g. NOR gates, the analysis can not detect the flip-flop, which leads to wrong truth tables. And moreover the result depends on the way the circuit is build.

And unfortunately a cmos fet also causes a cycle because of the bidirectional source and drain pins. 
This makes it impossible, e.g. to analyze a NAND built from cmos fet's.

The disadvantage is, that you can't demonstrate the the behavior of such a circuit by simply clicking the analyze button. 
But you can still add a test case to the circuit and run the tests instead, to demonstrate the behavior.

hneemann, 2018-05-31:
I forgot to mention: In the experimental mode, the cycle detector is disabled.

roy77, 2018-06-15:
I've been thinking about it. Wouldn't it be better to issue a warning for switches, CMOS and relays?
I find it very instructive when you can create a logic circuit from a CMOS circuit.

What is to be kept from activating the experiment mode via the settings that I can teach the students more easily.

Even better would be the improvement of feedback detection. If you indicate that there is a switchable connection between the two inouts of switch, CMOS and relay, the inouts in the analysis can be replaced by a short circuit or open. If more than one valid internal state (combinations of short circuit or open) is then possible for an initial sample, feedback is present.

hneemann, 2018-06-15:
> I find it very instructive when you can create a logic circuit from a CMOS circuit.

That's a good point! I will rethink the handling of switches by the cycle detector.