hneemann/Digital #157: Cannot open half adder

dmolony, 2018-06-02:
I built a half adder according to the example in the manual. When I select it from the custom menu, I get an error dialog:

Error importing the circuit half adder.dig!
Error importing the circuit /Users/me/Documents/half adder.dig!
caused by: Found a pin without a name.

All four pins have the correct names.

hneemann, 2018-06-02:
That's strange. 
Can you attach the circuit to the issue. Than i can dig into it.

dmolony, 2018-06-02:
[halfAdder.dig.txt](https://github.com/hneemann/Digital/files/2064950/halfAdder.dig.txt)
I don't know if this will work, I added .txt to the file name to bypass github's check.

dmolony, 2018-06-02:
The error message is also slightly wrong, it has left a folder out of the path. Perhaps it can't actually find the file?

hneemann, 2018-06-02:
Are you sure you sent the correct file?
It works as expected on my machine.

dmolony, 2018-06-02:
It's the only file I have, so it must be the correct one. I'm using macOS if that makes any difference. Also, see my previous message about the incorrect path.

dmolony, 2018-06-02:
I just tried quitting the program and restarting, and now it works. There's still a bug in there, but now I can't recreate it.

closed 2018-06-02 by dmolony

hneemann, 2018-06-03:
See also #33