hneemann/Digital #160: GAL16V8 limitations

periata, 2018-06-08:
I've never worked with the device before, but my understanding from reading the GAL16V8 datasheet is that it should be possible, depending on which of the 3 supported modes you're using, to use at least pins 13, 14, 17 and 18 as inputs.  Attempting to do this doesn't seem to work, though, causing an error "Pin 13 is not an input!"

hneemann, 2018-06-08:
You are right. The GAL16v8 allows to use the pins 13-19 as inputs. 
But Digital does not support that configuration. 
And this is mainly because I only needed a few inputs in my laboratory exercises. It was always enough to use the pins 2-9 as inputs. But the number of flip-flops was always close. Therefore, I did not sacrifice a macro cell for an input.
If you need more than the 8 Digital supported inputs, you can try WinCupl to create the JEDEC file.

closed 2018-06-11 by hneemann

xotmatrix, 2020-11-21:
Any change of heart on this? I was helping a friend convert a PAL to a GAL16V8 using Digital and a *.tru file I constructed but was disappointed I could not use pins 1 and 11 as inputs (allowed in "simple mode"), so JEDEC/CUPL output was impossible. She ended up using Digital's simplified equations in WinCUPL, so Digital was certainly helpful, and I thank you again for adding the plain text export feature for me.

hneemann, 2020-11-21:
To be honest, the 16v8/22v10 is completely out of my focus.
Although I like these chips very much, simply because they are so well documented and easy to understand, and therefore very suitable for a lab exercise, they have the problem that they do not support in system programming. This means that the students have to plug the chips back and forth between the circuit and the programmer, which is inconvenient for a lab exercise.

Therefore I have been using the ATF1504 for some time now. This chip is hardly documented which makes it impossible for me to create a JEDEC file directly. But there is a fitter available which can be integrated into Digital, so that a easy to use workflow could be implemented. But it supports ISP, which makes the lab exercises so much easier.