hneemann/Digital #161: Export to Verilog fails

darkstar007, 2018-06-14:
Hi,

I just tried the latest version (0.19) and loaded up the 'Processor.dig' file. This ran fine. When I then tried to export --> 'export to Verilog' option I got the attached error. I'm not sure if this is a bug or not, but the error message is not very clear.

Thanks,

Matt
[image]

hneemann, 2018-06-14:
The HDL export does not support high-z drivers, which are used in the simple Processor.
Please use the VHDLProcessor instead, which avoids the usage of such drivers and uses a muxer instead.
I will rename the VHDLProcessor.dig to HDLProcessor.dig and fix the error message.

darkstar007, 2018-06-14:
OK.

closed 2018-06-14 by hneemann

hneemann, 2019-11-07:
See also #184

commit 7d7a34d3 (2018-06-14): closes #161