hneemann/Digital #165: Built-in EEPROM DIPs share the same ROM data

bmelikant, 2018-06-22:
I'm not sure if this is an issue or a misunderstanding of how the built-in components work on my part. It seems that using the built-in 28C16 EEPROM package all of my ROMs share a single data instance. This doesn't appear to be the case for the generic EEPROM component. I would like to see the ability to use separate contents in each instance of the DIP EEPROMs if this is not already possible.

bmelikant, 2018-06-22:
I did find a temporary workaround for this issue. I decided to make separate sub-circuits for each of the ROMs I needed and simply copied the configuration from the built in EEPROM chip. While this works, I still think it would be easier to be able to insert several instances of the same circuit with different internal states.

hneemann, 2018-06-22:
It is possible to define different ROM contents in sub circuits which contain a ROM and are used multiple times:
Open your main circuit, than go to "Edit->Circuit Attributes->Advanced->Content of ROM's" to define the ROM content of each ROM.

closed 2018-06-22 by hneemann