hneemann/Digital #167: RAM/ROM contents

josepll, 2018-07-06:
It would be interesting if RAMs could have their contents preloaded from a file the same way ROMs are.
Similarly it would be interesting that ROM contents could be visualized during simulation the same way RAMs/RegFile can be. Even if the ROM content is not changing, it's very useful for debuging purposes.
--
On a side note congratulations for an excellent simulator. I'm planing on using it in my lab classes.
Right now I'm testing Digital with a 5-stage pipelined implementation of RISCV (RV32I subset only so far).
I'll post other issues that I found may improve the usability of Digital for academic pusposes.

hneemann, 2018-07-07:
But the natural behavior of a RAM is, that it has no defined content after start up.
Why not using a EEPROM instead, which has a predefined content?

josepll, 2018-07-08:
Usually for a simple 5-stage pipeline there are two memories: one for code and the other for data, However there are some data whose value is already initialized at program start. It is possible to use EEPROM, but there is no EEPROM with separated read/write ports, which is what most computer architecture books assume for the data memory stage.

josepll, 2018-07-09:
Another reason is debuging a design and making it to work. In a real processor it's the operating system that loads initiallized data into RAM. Whit a partially working processor we have no such luxuries. Having the OPTION to initialize from file will simplify things a lot. Don't make it mandatory, just keep the option for whoever considers it useful.

hneemann, 2018-07-09:
I will think about it.
In the past, I have used a ROM along with a RAM in a shared address space.
Therefore, some of the data found in the address space is predefined, even if it can not be changed.
Maybe this solution can help in your use case.

josepll, 2018-07-10:
Well, maybe is my particular use case, but since the design is already complicated (and I still have to add bypasses and interlocks)  i would rather keep the memories (see red circles) as simple as possible, since they are not the focus of the project.
[image]

hneemann, 2018-07-10:
I have decided not to implement such a feature for now.
There are two reasons for this decision: At first it's easy to use a ROM instead, to predefine the content of the address space (see example below which is a drop in replacement for the build in RAM). Secondly, it causes unnecessarily complex VHDL and Verilog templates to provide this function in VHDL and Verilog as well. Sorry for that.

[PreDefRAM.dig.zip](https://github.com/hneemann/Digital/files/2180237/PreDefRAM.dig.zip)

closed 2018-07-10 by hneemann

hneemann, 2018-07-10:
Maybe this circuit does the job.
Not perfect because of the AND gate in the clock line, but I expect it to work.

[CustomEEPROM.dig.zip](https://github.com/hneemann/Digital/files/2181968/CustomEEPROM.dig.zip)

josepll, 2018-07-11:
Thanks. I tried something similar. Maybe i end up writing on the negative edge to ensure i write within the cycle ensuring all signals are ready. I'll addapt one way or annother.

hneemann, 2018-07-23:
@josepll 
Would it be sufficient for your application if you could specify a filename in the circuit settings. The contents of this file would then be loaded into the RAM at startup.

josepll, 2018-07-23:
Definitely.

hneemann, 2018-07-23:
@josepll 
Please try the latest pre release [build](https://infdigital.dhbw-mosbach.de/).

josepll, 2018-07-31:
Thanks I was looking in the wrong place.

hneemann, 2018-07-31:
Never mind! :-)