hneemann/Digital #168: External components load from file

josepll, 2018-07-06:
I make extensive uses of external components and i see them as modules except they are implemented in verilog (or vhdl).

It would be interesting that they can be colored the same way as other modules to diferentiate them from basic components.

Also some of my components have several hundreds of verilog lines (and growing) Having a load/save feature so that an external file can be used would be great. That way more powerful editors can be used and the component can be separately tested with any external verylog (vhdl) compiler.

hneemann, 2018-07-07:
If a lot of external components are used, the simulation becomes very slow.
At which clock frequency did you end up?

closed 2018-07-07 by hneemann

josepll, 2018-07-08:
Several KHz, which is really good considering my students will advance clock slowly (probably by hand) to analyze the behavior of short instruction sequences.

Thanks for adding the load/save in latest prerelease. Will be a life saver.
Right now I use verilog to implement instruction decoder, alu and branch unit. In the future maybe cache controllers will also be implemented in verilog.