hneemann/Digital #170: Chronogram displays contents of multiwire

josepll, 2018-07-06:
Right now the measurement graph shows the level of multiwire signals. That is ok the only a handful of levels are possible. However for big busses (e.g. 32 bits) its much better to show contents and content changes. See example below.

[image]

closed 2018-07-07 by hneemann

hneemann, 2018-07-07:
@josepll 
Please try the latest [pre release build](https://infdigital.dhbw-mosbach.de/)

josepll, 2018-07-08:
Better, but it still is a bit confusing with high values.
See for instance what happens when increasing a 8 bits register (PC) by 4 every clock cycle
[image]
Changes are only shown every 2 cycles because levels don't change (visually) even though they are increased by 4.
For higher values (32 bit register) it's even worse.
[image]
In last pic register was still increasing by 4 every clock cycle.
Notice that levels are not very useful here.

hneemann, 2018-07-09:
@josepll 
Please try again!

josepll, 2018-07-09:
Now works great for small values (up to 8/12 bit)

However there are still issues when displaying big values (16/32 bit). See what happens when increasing register by 512.
[image]
0x400 gets displaced and 0x600 is missing
The issue appears because the visualization window gets clogged with big values that require extra space. I think the issue just requires extra zoom-in.
See what happens with a bit of zoom-out
[image]
I think will be enough to allow extra zoom so that the user can visualize what he needs.

hneemann, 2018-07-09:
@josepll
Now it's possible to zoom into the measurement graph.

josepll, 2018-07-09:
Excellent. Now 32 bit values are shown with no issue. Probably 64 bit values fit too.
Thanks a lot.

hneemann, 2018-07-09:
@josepll 
It was annoying to me that the names of the signals were not always visible. :-)
So I decided to reimplement the scaling and moving behavior of the data in the measurement graph.
If you have some time, please try the latest pre-release and let me know what you think about it.

josepll, 2018-07-10:
Only problem is that once you zoom-in the window remains in a fixed possition instead of advancing with the simulation. However if you remain at the default zoom the sliding window advances with the simulation as expectes, showing latest changes.

With default zoom (notice slider on the right)
[image]

With a bit of zoom at start (notice slider remains where it was before zooming in)
[image]

Same thing happens if you zoom out.

josepll, 2018-07-11:
So far works well. Hoever for complex designs/chronogrmas it gets a bit confussing. A proper chronogram is more intuitive.
[chronogram.pdf](https://github.com/hneemann/Digital/files/2183994/chronogram.pdf)

josepll, 2018-07-13:
That values don't jump up and down, that is very confusing and difficult to follow for big chronograms. Maybe the solution is to have the OPTION to show multiwire signals as levels or as bus contents
See the difference below:
Digital:
[image]
tkgate:
[image]
gtkwave:
[image]

For debugging complex circuits I prefer the later ones.

commit e9a414f4 (2018-07-07): adds numbers to measurement graph, if multibit values are displayed, closes #170