hneemann/Digital #172: Double-Throw switches and relays.

GandalfUK, 2018-07-16:
I have an interest in relay based CPUs but the current version of Digital has no double-throw switches or relays. I know I could create a custom circuit component using two single throw relays to produce the same function but this seems wasteful and wouldn't provide a proper circuit symbol. Double-throw switching is pretty fundamental in relay CPUs so I have attempted to add a double throw switch (SwitchDT) and relay (RelayDT) to Digital by duplicating and extending the existing single-throw switch and relay code. My attempt is here:
https://github.com/GandalfUK/Digital
I've replicated all the test cases for the original single-throw relay using the new double-throw relay and it all now seems to compile (mvn install) without error. I've only been looking at this over the last few days, so the code may be a little rough round the edges, (I guess the approach of duplicating and extending the existing relay code may not be the best way to tackle this either), but it seems to be working.
Would you be interested in a pull-request?

hneemann, 2018-07-17:
Adding a double-throw relay to build a relay-based CPU is an interesting idea. :-)
However, your implementation is quite complex because it does not reuse the existing switch. I do not want to add a modified copy of the switch code because this code is very complicated and difficult to test.
But I implemented a double-throw relay myself, which reuses the existing switch code.
This minimizes the amount of code and the required testing of the new component.
I removed the "normally closed" option because it does not make sense with a double-throw relay, and I also added a "pole number" option to allow multi-pole relays.
Here you can find the latest [pre-release build](https://infdigital.dhbw-mosbach.de/).
Let me know what you think about it.

GandalfUK, 2018-07-17:
Yep, when I looked at the existing code for the relay and found that it was based on the switch, I was torn between just implementing it as two single-pole switches as you have done, or to create a new double-pole switch and base the DT relay on that. I noticed some of your earlier comments about the switches/relays being expensive to simulate, so I thought that just doubling up on the switches might be the wrong approach, I also thought that a double-pole manual switch might be useful in its own right, so I ended up trying to create a DT switch by copying and extending the original switch code, then creating 3 "optimised" internal models (fan-in, fan-out, and mixed). By the end it did get more complicated that I originally thought! :-) (although it did seem to work and pass all the test cases I could think of).

I think your approach is probably better, certainly much simpler. Do you think it will suffer from any performance issues? (e.g. if tying to simulate hundreds of relays)?
Also I still think a manual double-throw switch would be useful, do you have any plans to add this?

I was about to start work on the addition of multiple relay poles! :-) so thanks for that addition. I had originally hoped that I could use the bus splitter/combiner for the poles, but I  discovered that they are directional and so I couldn't see a way to use busses with the relays (as there is no way to split and combine them while maintaining their bidirecionality). Do you have any thoughts on this? Is there a way to split/combine bi-directional busses at the moment? Would the addition of a bi-directional splitter/combiner be necessary?, (and if so do you have any plans to add one)?

One last thing is that most relay CPU designs make use of "floating 0" as this avoids *many* problems (esp. shot circuit issues). e.g.:
http://web.cecs.pdx.edu/~harry/Relay/VideoTutorial/SupportingFiles/Slides/Slides1up.pdf
http://www.electronixandmore.com/projects/relaycomputertwo/index.html
When the current input IO pad is set to tri-state it cycles between high-z, 0, and 1, which makes experimenting with "floating 0" circuits cumbersome, (the 0 is not a valid input in "floating 0" circuits and usually causes a short-circuit). In my last commit I have modified the input pad to have a "floating 0" option that simply skips the 0 if selected, (i.e. a tri-state input pad with floating 0 would only cycle between HighZ and 1). I think this is a useful addition when working with relay circuits.

GandalfUK, 2018-07-17:
I've just been playing with your RelayDT in my existing circuits, and while I noticed your comment earlier about not including a "normally closed" option, I didn't realise how problematic this would be. Having some relays "normally closed" can make the wiring much cleaner in many circuits, (since you can choose which of the throws are normally-open and which are normally closed without rotating the relay. Is there any chance you could add this useful option back in?
On a similar note, some circuit layouts can be quite messy because there is no way to "mirror" the components, (this is particularly problematic with relay circuits). Do you have any plans to implement a "mirror" option (in addition to the already existing rotate)?

hneemann, 2018-07-17:
> I think your approach is probably better, certainly much simpler. Do you think it will suffer from any performance issues? (e.g. if tying to simulate hundreds of relays)?

I'm not sure. A switch is much more expensive to simulate than a bus driver. But if the frequency drops to 100Hz, compared to the 100kHz, which are possible if a processor is built without a switch, this may be ok in this case.

> Also I still think a manual double-throw switch would be useful, do you have any plans to add this?

Maybe. But you can easily connect a input to a relay to get a "manual" switch. Not sure if someone needs it.

> I had originally hoped that I could use the bus splitter/combiner for the poles, but I discovered that they are directional and so I couldn't see a way to use busses with the relays (as there is no [...]

Every signal in Digital is a bus. But some busses are only one bit wide. Thus also a relay can switch many bits with a single pole. Maybe that helps building a processor. There is also a bidirectional splitter, but this is more of a hack because it has an input which defines the direction.

> When the current input IO pad is set to tri-state it cycles between high-z, 0, and 1, which makes experimenting with "floating 0" circuits cumbersome, (the 0 is not a valid input in "floating 0" [...]

I will think about it.

hneemann, 2018-07-17:
> I've just been playing with your RelayDT in my existing circuits, and while I noticed your comment earlier about not including a "normally closed" option, I didn't realise how problematic this would [...]

All you have to do is to swap the two wires. Can that be a problem? But to be honest, the most complex relay circuit I've ever built is a 2-bit adder.

> On a similar note, some circuit layouts can be quite messy because there is no way to "mirror" the components, (this is particularly problematic with relay circuits). Do you have any plans to implement a "mirror" option (in addition to the already existing rotate)?

No! I don't like a mirror option. :-)
But i can understand that some people do. I have to think about it.

hneemann, 2018-07-18:
> If you aren't going to work on a mirror option then let me know. I have a few ideas about how to implement this with only minor changes to the existing code (e.g. by extending the rotate transform), or with slightly larger changes (e.g. by adding a new mirror key and a matching mirror flag to both the translate and rotate transforms).

The mirroring itself is a no brainer. That's not a problem. In the moment I'm working on an SVG import branch where i have already implemented real transformations (a 2x2 matrix and a offset vector) which makes if even easier.

The problem is the text positioning which currently works with only two anchor points. This works great if there is no mirroring, but if you add mirroring you have to add more information to get the text position right and that means i have to touch all the shapes.

Try to rotate a shape: The text does not rotate in the same way as the shape does. This feature needs to be maintained if a mirror option is added.

GandalfUK, 2018-07-18:
Ahh I see! I hadn't thought about the text rendering, I may take a closer look later.

In the meantime I have just pushed a commit to my repo that changes the layout of your RelayDT to better match the existing single-throw relay. i.e. The common pin (and normally-open pin) is now in the same place on both relays. You now don't have to re-wire your circuits so much when swapping the ST and DT relays, which makes experimenting with different relay circuits much more convenient. (I've also changed the examples and test cases to match the new pin layout so there are no errors when building). I can create a pull request if you are interested.

Do you have any plans to add the multi-pole option to the existing single-throw relay (and perhaps the switch too)? Since the existing bus spiltters can't be used in relay circuits I now find I need these, (If you don't already have this in the works I could add it in later tonight).

GandalfUK, 2018-07-18:
I noticed that you created 2 full adder test circuits for the double-throw relay. If you have an interest in such things, I have just created a new test circuit implementing Konrad Zuse’s full adder relay circuit. I believe he invented this for the Z3 computer built in 1941. It's particularly interesting as it only requires 2 relays (using "floating 0" logic).

GandalfUK, 2018-07-18:
As mentioned above, I found myself in need of multi-pole single-throw relays this evening due to the lack of bi-directional busses, so I've just pushed a commit to my repo adding this, (I've followed your pattern in the RelayDT to do this). Please take a look if you are interested, I can create a PR if you want.

hneemann, 2018-07-19:
> As mentioned above, I found myself in need of multi-pole single-throw relays this evening due to the lack of bi-directional busses, so I've just pushed a commit to my repo adding this, (I've [...]

A PR is not necessary. I have already pulled your latest commit.

GandalfUK, 2018-07-19:
> You're right: It can not be done exactly as Harry Porter did. But it can be done slightly different. But maybe this solution is not as relayish and elegant as it could be. :-)

Yes, I looked at options like that, but the simulation circuits become something very different from what you would build in the real-world, and my goal here is to simulate something that I intend to build, (to get the bugs out before I start building it! :-) ). e.g. this is a typical 4-bit relay register you might actually build:
[image]
Ideally I would like to reduce the Select relay to a "single-pole" 4-bit bus, (which simplifies the wiring layout later), but as you have found, the only way to do this is to introduce extra "direction" lines that wouldn't be present in the real-world. If you then try to build a CPU with these modified registers, the instruction-decode/control logic must be very different in the simulation than in the real world (to handle all these extra "direcion selector" lines).

It's not a major blocker for me at the moment as I can just draw out the bus-lines separately, but (like the floating-0 option for the input pads), it makes simulating "real-world" relay circuits much more convenient.

hneemann, 2018-07-20:
That sounds like an interesting project. Is there already some documentation?

hneemann, 2018-07-24:
See #177

closed 2018-07-24 by hneemann