hneemann/Digital #174: Please implement assembler API interface for EEROM and RAM and a load data option for RAM.

sgparry, 2018-07-21:
The LMC CPU architecture I am recreating in digital has no ROM - I am implementing an assembler I would like to be able to load programs and debug them. Please add the assembler API that is available for ROM to EEROM and RAM and the ability to load programs to RAM. Yes, I know having a RAM load option is not very realistic but LMC is too primitive to even implement its own bootloader.

hneemann, 2018-07-21:
Why not putting a ROM and a RAM together to a shared address space? This would allow to use the existing interface.

hneemann, 2018-07-22:
See also #167.

sgparry, 2018-07-22:
Not sure how would I arbitrate between the two chips?

hneemann, 2018-07-22:
With the latest commit you can also access a RAM from the assembler API and write code to it.

And [here](https://github.com/hneemann/Digital/blob/master/src/main/dig/processor/core/Memory.dig) you can find a combined memory. The Von Neumann processor example uses it. 

[image]

sgparry, 2018-07-22:
Thank you again as always Helmut  Having the RAM assembler accessible and loadable I think will solve this requirement. I can add calls to the assembler api into my LMC assembler allowing my students to create, load and debug programs.
I cannot use the shared address space approach - The Binary LMC only has a 5 bit address space and no assembler directives for setting an address! However I could create a hardware bootloader to load from a ROM. It opens useful discussions with the class on loaders and bootloaders.
Another possibility would be if EEROM was assembler accessible, but I have the feeling that might prove more difficult than RAM?

hneemann, 2018-07-23:
> Another possibility would be if EEROM was assembler accessible, but I have the feeling that might prove more difficult than RAM?

EEPROM is already accessible by a assembly IDE.

closed 2018-07-23 by hneemann

sgparry, 2018-07-23:
Thanks Helmut. You move this project so fast I just can't keep up! :-)

commit 399d7c54 (2018-07-23): Allows the pre-loading of program memory if a RAM is used as such. See #167, Closes #174