hneemann/Digital #177: Refactoring of relays and switches

hneemann, 2018-07-24:
Refactoring of relays and switches to allow multi pol switches and to make switches and relays share the same code.

closed 2018-11-03 by hneemann

hneemann, 2018-11-03:
@GandalfUK 
I suspect you are the person who has build the most complex relay circuits.
I would be glad if you could test the last [pre-release](https://infdigital.dhbw-mosbach.de/) build on occasion.

GandalfUK, 2019-01-14:
Hi.

Sorry about the lack of reply for so long, I had to put my relay-cpu project on hold for a while.

I took a look at the latest pre-release (0.21-11), but it still has a number of issues for me (or I think for anyone trying to work with relay logic):

  1.  The lack of an option to make the input pad flip only between logic 1 and High-Z is a real usability killer. As I mentioned a while back, “floating 0” logic is the only practical way to build large relay circuits, and to work with such circuits in Digital you need an input pad that can flip between logic-1 and high-Z (and back again) without transitioning through logic-0, (which triggers errors in such circuits). I know there are ways to work around this with additional relays, but the modification I made to enable “floating-0” input pads a while ago was pretty small, (just an extra tick -box on the input pad config page and an extra conditional behaviour in the code), and it enhances the usability considerably.

  2.  The lack of a true bi-directional bus/splitter/combiner is still a real problem. (I notice that others have made similar comments in the last few months). While not a complete blocker, it makes relay CPU circuits *much* more cumbersome to work with as all busses must be drawn out in full.

  3.  The double throw relay (and switch) still have no option to change their default (off) state (the single throw versions do have this option). This means you often have to do cumbersome re-wiring in situations where this is needed, (this is especially painful when experimenting with circuits where you are swapping out single and double throw switches).

  4.  The lack of a mirror option is also still problematic for double-throw switches and relays, (and I suspect other components that are not symmetrical under reflection). There are circuits that are just much simpler to draw with some relays where the coil is on top and the multiple poles are on the left.

I “fixed” problem 1 and 3 in my fork some time ago, but I never got round to dealing with 2, and 4 (item 2 being my main remaining concern). If you aren’t planning to address these items in the future, I’ll try to get some time in the next few months to come back to my fork and fix item 2 at least.

Regards,
Craig.

commit 10b97829 (2018-11-03): refactoring of relays and switches, closes #177