hneemann/Digital #179: [BUG] Could not write values from the external process; Timeout reading data from external process

amreo, 2018-07-29:
When I try to run my project I receive this error "Error creating the circuit. Could not write values from the external process! Affected are: External, Signal o. Caused by: Timeout reading data from external process!".

I created a simple project with a button, a "external" (with GHDL) and a led:
[button]->[external]->[led]
In the extern I insert this code:
```
library ieee;
use ieee.std_logic_1164.all;

--------------------------------------------------

entity wire is
port(	i: in std_logic;
	o: out std_logic
);
end wire;  

--------------------------------------------------

architecture behav of wire is
begin

    o <= i;

end behav;

--------------------------------------------------
```
Full error:
```
de.neemann.digital.core.NodeException: Could not write values from the external process!
Affected are: External, Signal o.
	at de.neemann.digital.core.extern.External.writeOutputs(External.java:90)
	at de.neemann.digital.core.Model.doMicroStep(Model.java:288)
	at de.neemann.digital.core.Model.doStep(Model.java:257)
	at de.neemann.digital.core.Model.init(Model.java:179)
	at de.neemann.digital.core.Model.init(Model.java:166)
	at de.neemann.digital.gui.Main.createAndStartModel(Main.java:1281)
	at de.neemann.digital.gui.Main.access$3900(Main.java:86)
	at de.neemann.digital.gui.Main$RunModelState.enter(Main.java:1619)
	at de.neemann.digital.gui.Main$RunModelState.enter(Main.java:1611)
	at de.neemann.digital.gui.state.State$1.actionPerformed(State.java:82)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.io.IOException: Timeout reading data from external process!
        
	at de.neemann.digital.core.extern.handler.StdIOInterface.readLine(StdIOInterface.java:115)
	at de.neemann.digital.core.extern.handler.StdIOInterface.readValues(StdIOInterface.java:157)
	at de.neemann.digital.core.extern.External.writeOutputs(External.java:88)
	... 46 more

```
Digital version: 0.19
GHDL version: 0.36dev
OS: Archlinux
Java: ```openjdk version "1.8.0_172"
OpenJDK Runtime Environment (build 1.8.0_172-b11)
OpenJDK 64-Bit Server VM (build 25.172-b11, mixed mode)```

hneemann, 2018-07-29:
Could you please try GHDL 0.34?
It seems to me a problem with the new --UNBUFFERED option in GHDL.

hneemann, 2018-07-29:
I have just tested your VHDL code with GHDL 0.36dev.
It works just fine on my machine.

amreo, 2018-07-29:
Sorry. I'm failing to build the 0.34 version of GHDL.

amreo, 2018-07-30:
Version: e8f46c43e37267363ae010573aa3a3b361e8a532
I tried to debug the programs and I found out that:
* The built wire seems to work (executed manually). When I write 1 it responds with digital:1 and when I write 0 it respondes with digital:0.
* process in StdIOInterface seems to work (It is alive)
* process.getInputStream().available() is always 0 (and it's strange)
* process.getErrorStream().available() is always 0.
* The condition in the while(line 80) in setReaderWriter is never true
* I suspect there is a bug in the JVM. Today I try to use the java process API

hneemann, 2018-07-30:
It seems to me a problem with the Linux IO buffering. 
Can you try to add in the file ApplicationGHDL.java in line 33 then option ["--unbuffered"](https://ghdl.readthedocs.io/en/latest/using/Simulation.html) to the ghdl command?

New line 33:

`ProcessBuilder pb = new ProcessBuilder(ghdl, "-r", "--unbuffered", "--ieee=synopsys", ...
`

amreo, 2018-07-30:
Yes. It works when I added the --unbuffered options, but at the end of the command: 
`ProcessBuilder pb = new ProcessBuilder(ghdl, "-r", "--ieee=synopsys", "stdIOInterface", "--unbuffered").redirectErrorStream(true).directory(file.getParentFile());`

commit 70eda1fc (2018-07-30): fixes an ghdl io buffering issue, closes #179
closed 2018-07-30 by hneemann