hneemann/Digital #180: Memory Monitor

sgparry, 2018-07-29:
It would be useful to have a simple component that would allow the hex, decimal or binary contents of a RAM, ROM or EEPROM chip to be displayed during simulation. I would be willing to code.

hneemann, 2018-07-30:
But there is already a RAM content monitor dialog which also works for the EEPROM components. 
Do you want to improve the available dialog?

sgparry, 2018-07-30:
I missed the contents dialog at first, which I think is indicative of the problem. What would be helpful is if it were possible to either have a setting on a memory component that makes its mem contents dialog come up automatically when the simluation starts, or better still be able to have a component that shows the same content but in the circuit, like the probe does. That way you can position it where it relates to the memory being inspected.

hneemann, 2018-07-30:
Hmm! I have to think about it.
I decided to implement a separate dialog instead of a special shape because there is a lot of data which is difficult to inspect in such a small area.

If the memory is located inside a embedded circuit, you can inspect it by using the measurement table.

sgparry, 2018-07-30:
On 30 July 2018 14:22:22 BST, Helmut Neemann <notifications@github.com> wrote:
>Hmm! I have to think about it.
>I decided to implement a separate dialog instead of a special shape
>because there is a lot of data which is difficult to inspect in such a
>small area.
>
>If the memory is located inside a embedded circuit, you can inspect it
>by using the measurement table.
Is there a way of inspecting the actual memory contents using the measurement table? I have only been able to view data / addresses going in and out by adding probes.

hneemann, 2018-07-30:
The measurement table dialog has a menu, which allows to inspect the memory components contained in the model.

sgparry, 2018-07-30:
Ah thanks! Very cool.

closed 2018-07-30 by hneemann