hneemann/Digital #181: Unable to write to EEPROM

LarsiParsii, 2018-08-15:
Hi there,
First of all, I really appreciate the work you've put into this program, I've used it together with Crash Course's series on computer science. I've made a somewhat working CPU, with simple arithmetic functions, and ability to store the output from those functions.

I've ran into a problem however when trying to write data (the sum of the addition) to the EEPROM (used since I need the data after the program have been terminated). The CPU proceeds through the steps with the "clk" button. At the fifth step, it sends the value (3/0b11) from register A through the bus, to the EEPROM, while also setting "WE" pin on the EEPROM to high, making it writable. However, when I open the EEPROM to see if it had been saved in address "0", there's nothing to be seen.

I've attached the file for the "CPU" [**here**](https://drive.google.com/file/d/1jcnBIQwaxJTf-Vi-EZL-2KnSGCfa4ZMF/view?usp=sharing). If you would have the time to take a glance upon my project, it would be greatly appreciated! If you need any more information, just ask! :)

hneemann, 2018-08-15:
There are some files missing. I'm not able to run your circuit.
Please use the "Export->Export to ZIP" function to create a ZIP file which contains all the necessary files.

LarsiParsii, 2018-08-15:
[CPU_eng.zip](https://github.com/hneemann/Digital/files/2290332/CPU_eng.zip)

Here it is. How silly of me to forget the other circuits. By the way, I'm not a specialist when it comes to designing circuits, so some circuitry may seem unnecessary.

hneemann, 2018-08-15:
It seems to me, your circuit works well. But of course I'm not sure what the intended behavior is.
But I noticed that the value to be written changes again after WE changed to high. That could be the problem. The memory components work as follows: The value to be stored is read at the moment of the rising edge on WE. Changes that still take place after the rising edge are ignored. Even if the changes take place only one gate delay after the rising edge of WE.

LarsiParsii, 2018-08-15:
Thanks for the quick answer! I added a delay to the WE pin, and now it's works! However, I wonder why I can't immediately send other data over the data bus, right after writing to the EEPROM.  If you try to insert the operation for copying from register A to RAM-address 0001 (0b01000001 / 0x41) right after the previous one (0x40) you will see that the EEPROM ignores that data. This requires me to have an "empty" instruction after every STORE operation, allowing the EEPROM to store the data before trying to access new one.

hneemann, 2018-08-15:
Inserting a delay gate is not a good idea as it is very fragile. It may be better to reverse the WE polarity so that the data is stored on the falling edge. With the rising edge of the clock, all data is generated and placed on the data bus, which can last some gate delays, and the subsequent falling edge of the clock then stores the data.

closed 2018-08-15 by hneemann

LarsiParsii, 2018-08-15:
I tried reversing the polarity, but that enabled the write pin at all times, when I just want it to write when wanting to store from register A to the EEPROM.

hneemann, 2018-08-16:
Data is only written on a rising edge of WE (or a falling edge if WE is inverted). If there are edges on WE where you do not want to write data, you must turn off CS.

If you want to compute and store data in the same clock cycle, you can not do it on the same clock edge because the data is not ready when the edge occurs. This means that the data is not present at the data input of the register / RAM / EEPROM.
So the idea is: Prepare the value at the rising edge and store the prepared value at the falling edge.

LarsiParsii, 2018-08-17:
Hello again! While I managed to write data to the EEPROM, I've ran into another problem.

The number 6, 7, 8 and 9 is copied from the EEPROM to the registers respectively.

Then, the data is supposed to be written back to the EEPROM in the reverse order. First, data (the number 9) gets written from register D (the one furthest to the right) to the EEPROM's first address (0). 
The next step is to write the data from register C (the number 8), to the EEPROM's second address (1).

However, when this clock cycle is finished, the data from the previous register is used, writing "9" instead of "8" at address 1. I've tried changing the clock pin so that it writes at the falling-edge. But to no avail. I'm not pushing this problem onto you, but if you have the time to help me figure out where I've gone wrong, that'd be greatly appreciated, as I love your simulator! <3

Here's the updated CPU, with OP-codes and the program hex-file:
[CPU.zip](https://github.com/hneemann/Digital/files/2297660/CPU.zip)

(Btw, I've been streaming myself tweaking and building this project [here](https://www.twitch.tv/larsiparsii)

hneemann, 2018-08-17:
First of all: Your CPU is quite complex. There are a lot of diodes which are not necessary. In my opinion a circuit is easier to understand using an OR gate instead of a distributed wired OR build with diodes. And if you don't need an OR gate, don't insert diodes.
And it's may be easier to use a ROM to build a DECODER circuit. The circuit is much more flexible and becomes less error prone doing so.
And why there are two clock signals? A circuit with multiple async. clock domains is much more difficult to build because you have to think about how to cross the clock domains.

The processor contained in the examples is build from 58 components and uses 16 registers and is able to decode 64 instructions. Your processor has four registers and is build from more then 100 components. Try to keep it simple. The fewer components are used, the easier it is to understand the circuit.

If your problem only affects the first instruction, than it's maybe a problem of polarity. If the circuit is started the clock signal is low. The first instruction is output from the ROM and all the control signals are generated and ready to use. In this situation a rising edge at the clock can store a value to the EEPROM and the falling edge of the clock should increase the program counter. Maybe you should store all the data at a rising edge and increment the program counter with a falling edge. Otherwise your first instruction is skipped because the first rising edge at the clock increases the program counter without storing something.