hneemann/Digital #186: IEEE Shapes not really to the standard

mgriebling, 2018-09-13:
First of all, I'm very impressed with **_Digital_** and would like to adopt it for use in some of my work.

I've updated the IEEE shapes to look closer to the standard shapes.   See below.  

[image]

Since my changes necessitated changing the width of the shapes, I also adjusted the circuit builder.

[image]

I also made some changes to allow the exported image/svg files to not show the connection dots which are unnecessary and look ugly ;-)   I am trying to get good quality circuits for publication in a book I'm working on.  Here's an example without the dots.

[image]

I know you probably don't want to include my changes in your Digital implementation because they would break many diagrams due to the component width changes but I would still be very interested in keeping up-to-date with your Digital implementation.  Perhaps, I should create a plug-in?  Not sure how to do this, but this may not use my shapes for circuit builder.  Suggestions for how to proceed would be appreciated.

BTW, I would also be interested in producing a custom generator/simulator to allow C-style code generation for inclusion in some hardware I'm also designing.  Then circuits could be simulated in C on an embedded micro without necessarily having a computer (or an expensive FPGA).  I don't know if anyone has been working in this area.

hneemann, 2018-09-13:
> I am trying to get good quality circuits for publication in a book I'm working on. Here's an example without the dots.

Try the LaTeX export. With this export you get the best quality. You can create SVG Graphics where the Text in the shapes is generated by LaTeX. This way the graphics use the same font which is used for the normal text. And the dots are also removed. :-)

> I know you probably don't want to include my changes in your Digital implementation because they would break many diagrams ...

The problem is that there are two types of [diagrams](https://en.wikipedia.org/wiki/Logic_gate). The symbols that are more commonly used in the US and the symbols that are mostly used in Europe or at least in Germany. And circuits should work with both types of diagrams. And that means the pin positions have to be the same. That's why the US symbols look a bit compressed.

> BTW, I would also be interested in producing a custom generator/simulator to allow C-style code generation for inclusion in some hardware I'm also designing. Then circuits could be simulated in C on [...]

That's an interesting idea! As far as I know nobody works on it. But [FPGAs](https://tinyfpga.com/) are not that expensive anymore.

mgriebling, 2018-09-13:
> Try the LaTeX export. With this export you get the best quality. You can create SVG Graphics where the Text in the shapes is generated by LaTeX. This way the graphics use the same font which is used for the normal text. And the dots are also removed. :-)

Unfortunately, I am doing the book in iBooks so there is minimal LaTex support (just equations).  PNG graphics are about the only way to incorporate diagrams.  I will be incorporating a widget to simulate the circuit within the book.  The program to create the simulator also prefers PNG graphics.

Perhaps a preference setting flag to disable dots for exported outputs would be nice.  BTW, I used the LaTex flag to disable the dots in the other outputs.

> The problem is that there are two types of [diagrams](https://en.wikipedia.org/wiki/Logic_gate). The symbols that are more commonly used in the US and the symbols that are mostly used in Europe or [...]

I took that into account and adjusted the Euro symbols to be wider.

> That's an interesting idea! As far as I know nobody works on it. But [FPGAs](https://tinyfpga.com/) are not that expensive anymore.

They are fairly expensive for large gate counts if you want to simulate anything serious.  I can probably fit 100 circuits (some would be fairly large) into a $5 micro without having to reprogram fuses.  For teaching purposes, no one really needs GHz circuits -- especially with an introductory course.  Most FPGA software is also horrendous to use.  Your approach is much nicer.

nevercast, 2018-09-13:
I would be seriously interested in a microcontroller export option. Something Atmel targeted would be excellent. Here probably isn't the place to start a thread on it though, perhaps raise another issue to discuss that if you are serious and believe you've the time to implement that, @mgriebling

mgriebling, 2018-09-13:
> I would be seriously interested in a microcontroller export option. Something Atmel targeted would be excellent. Here probably isn't the place to start a thread on it though, perhaps raise another issue to discuss that if you are serious and believe you've the time to implement that, @mgriebling

I was thinking about Microchip micros but using C as an output format should be flexible and more or less independent of the target micro.  I'll start another thread once I've looked at Digital's structure and can evaluate how much work it would be.

hneemann, 2018-09-13:
It's not very difficult to create a c-code export. All you need to do is to implement a few lines of code and some c-code templates.

But to be honest: It feels wrong to me to use a simulator on a PC to create a new simulator which is able to run on a micro. Why not using the PC simulator?

> They are fairly expensive for large gate counts if you want to simulate anything serious. I can probably fit 100 circuits (some would be fairly large) into a $5 micro without having to reprogram fuses.

A Mimas board costs about $50. Not much more than a micro board which is suitable for a lab exercise. And such a board is able to run a processor including vga video output, which is nice to use in a processor architecture course.

> For teaching purposes, no one really needs GHz circuits -- especially with an introductory course. 

See above. Think about creating a 16 bit processor in Digital and than create a c-code simulator which runs the simulated processor on a 8 bit micro. :-)

> Most FPGA software is also horrendous to use. Your approach is much nicer.

You are right: But with Digital you can create a Basys3 export including a proper project file. After that all you have to do is to open the file with Vivado and click on "Generate Bitstream". After a while you can click on "Program Device". That's it.

And if you only want to use some simple logic in an introductory course, go for the ATF1504.
Very easy to use, fast workflow and cheap.

Do not be angry with me, but creating c-code to use a microprocessor-based simulation in a lab exercise feels like I'm cheating the students.

hneemann, 2018-09-13:
> Not everyone is blessed with a PC.

I am not sure if I understand you correctly! In any case, you need access to a PC to use Digital.

> As you say Digital can provide different outputs quite easily so no point in limiting people to just one.

Thats a good point! :-)
I will think about creating c-code.

nevercast, 2018-09-14:
The use-case for me, is that I have many atmel/espressif SoCs about my work place and home, If I could deploy my digital circuit to one of these devices that I already have, and see that my breadboard circuit works with my real world hardware (considering debouncing, etc), this would give me confidence in my digital design to purchase an FPGA.

While FPGAs are very very cheap now and they no longer require large programmers, I don't have FPGAs in my draw, I do have micros. Its just another option to test a design.

hneemann, 2018-09-14:
@nevercast 
But what do you gain if you test your circuit on a slower simulator compared to testing it on a faster one? The simulation algorithm is mostly the same, just running on a micro. And testing itself is much more difficult on a micro.

Debouncing is a perfect example: How do you test such a algorithm on a simulation running on a micro? Connecting a button and pressing it a few times? Not a very reliable method. And even if it works: What makes you believe that it also would work on an FPGA running at much higher speed?

In the PC simulation you can apply numerous stimuli and ensure the logic works as expected. And in terms of timing, you can not learn much more from a simulation running on a micro.

In my opinion does a test on a micro add more work, but it does not give you more trust.

mgriebling, 2018-09-14:
My original use case for the C code generation was to deploy an inexpensive way to simulate many circuits (i.e., 100) without requiring a computer to program the test bed.  These circuits would be preloaded and interfaced to an inexpensive keypad and LEDs.  The keypad would let you select a circuit to simulate and then keypresses would be used as inputs to that circuit.  Outputs are observed on the LEDs.  Of course, this is a functional simulation only.  Debounce would be taken care off by the keypad interface.  You can host this on a micro costing a few dollars, storing 128KB and having speeds of 64MHz.

hneemann, 2018-09-14:
@mgriebling 
Now I got it!
There are e.g. 7 DIP switches that select the function: If they are set to e.g. 56, the circuit behaves like a jk flip-flop. And selecting, e.g. 45 at the dip switches causes the circuit to behave like a 4-bit adder. No need to reprogram the circuit.
While students can not design their circuits themselves, they can at least experience them.
What is the most complex circuit you think about?

hneemann, 2018-09-14:
Or maybe there are only the components of a processor like the ALU, the registers, the RAM or the CU.
So students can connect multiple boards together to build a CPU themselves.

hneemann, 2018-09-15:
I decided not to change the US forms. You can not make both types of shapes perfectly compliant. And the decision made in my opinion is the best compromise.

mgriebling, 2018-09-15:
How would I create an official branch off of **Digital**, perhaps renamed to avoid confusion, so that I can use my symbols and retain future changes that may be made in **Digital**?  I've attached my changes in accordance with the GPL and in case anyone would like them.  BTW, the attached changes include changes to the Euro symbols so they are interchangeable with the ANSI/North American symbols.  Note: I haven't extensively used Github yet, so "branching" is a new process for me.
[IEEESymbolChanges.zip](https://github.com/hneemann/Digital/files/2385554/IEEESymbolChanges.zip)

hneemann, 2018-09-15:
At first create your own fork. 
In your local clone you can add my repository as a new "remote":
```
git remote add hneemann https://github.com/hneemann/Digital.git
```
And then you can pull changes I have made to your local repository:
```
git pull hneemann
```

mgriebling, 2018-09-15:
Thanks, I'll try that.

hneemann, 2018-09-29:
@mgriebling 
I have added wider shapes without breaking the compatibility to older circuits and the examples. (See branch wideShapes)
Please let me know what you think about it.

it should be possible to open one of your new circuits, select all and set the "Wide Shape" check box in all gates.

mgriebling, 2018-09-29:
Excellent news, Helmut. I've tried it out and everything seems to work without any problems.  After I wrote out the modified circuit, I can still read it in my program as well.  

Would you be adding a global preference switch to enable the wider option?

Here's the before with one of my diagrams opened with Digital:

[image]

And here it is after setting the wideShapes switch still in Digital:

[image]

And here is the same diagram opened from LogicBLOX after the switch was set:

[image]

hneemann, 2018-09-29:
> Would you be adding a global preference switch to enable the wider option?

It's not possible to add such an option because that causes a lot of compatibility issues.
(A global setting needs to match a certain circuit to make the circuit work.)

Buts it's possible to add a global setting which selects the "wideShape" check box by default, if a new circuit is created.

mgriebling, 2018-09-29:
By "global option" I just meant that the wider circuit components would be shown in the block selection views and the"wideShape" flag for that diagram would be set.  I suppose if an older diagram were loaded it could still show the narrow gates.  Then, by setting the "wideShape" flag for the older diagram, it could be edited to convert it to the wideShape format.  That's more or less what I've been doing when editing your existing circuits and adapting them for the wider gates.  BTW, if you would like the converted diagrams (supporting the wider shapes), I can zip them for you to use in a separate wide-gate library.  All these diagrams would still need to have the "wideShape" flag set though.

Once you release the "wideShape" flag, I don't mind editing my converted diagrams to use this flag.