hneemann/Digital #187: [enchancement] flip (mirror) components

chriscamacho, 2018-09-14:
As well as rotating components, flipping (or mirroring) would be very useful (I think with rotation only flipping one axis would be needed ???)

This is especially noticeable when using some 74 series "chips" with _Splitters_ as some inputs or outputs end up the wrong way round 7-0 matching 0-7 for example on the chip, while you can rotate then the other side of the _Splitter_ is in the way of one pin of the other side... (hope that makes sense!)

hopefully flipping can implemented at a low level in the object hierarchy and can be implemented in one place affecting many (or even all) types of components

hneemann, 2018-09-14:
See #172 [Detail](https://github.com/hneemann/Digital/issues/172#issuecomment-405820540).

hneemann, 2018-09-14:
See also this questions: #70, #141 
This maybe helps with your splitter use case.

chriscamacho, 2018-09-14:
doh! #70 and it dawned on me.... maybe one for the FAQ its easy to miss you can reverse the order... (so not needing flipping)

but... is this the correct syntax ???

`7-7,6-6,5-5,4-4,3-3,2-2,1-1,0-0`

chriscamacho, 2018-09-14:
just a thought... adding a period could be used as short hand for a single pin so :
`7-7,6-6,5-5,4-4,3-3,2-2,1-1,0-0`
could be more conveniently written as....
`7.,6.,5.,4.,3.,2.,1.,0.`

closed 2018-09-15 by hneemann