hneemann/Digital #188: 74x182 circuit

chriscamacho, 2018-09-14:
[74182.dig.txt](https://github.com/hneemann/Digital/files/2383646/74182.dig.txt)
remove .txt from end of file, github complains about .dig ....

this is a look ahead carry generator, which can be used with the 74x382 I sent you earlier, I was going to write a test case but the only truth table I could find included lots of blanks in each line as well as X's and I'm unsure how to handle that in the test case ?

chriscamacho, 2018-09-14:
ok it passes this test but says too many entries, does that mean its skipped some tests??

markdown doesn't like the comment char (markdown for <HR> ??)

had to remove the GND and VCC pins connected to a power component before it would run the test, not sure what I'm doing wrong...

> C_n ~G0 ~P0 ~G1 ~P1 ~G2 ~P2 ~G3 ~P3 C_n_+_x C_n_+_y C_n_+_z ~G ~P
> `#`
> X   1   1   X   X   X   X   X   X   0       X       X       X  X
> 0   1   X   X   X   X   X   X   X   0       X       X       X  X
> X   0   X   X   X   X   X   X   X   1       X       X       X  X
> 1   X   0   X   X   X   X   X   X   1       X       X       X  X
> `#`
> X   X   X   1   1   X   X   X   X   X       0       X       X  X
> X   1   1   1   X   X   X   X   X   X       0       X       X  X
> 0   1   X   1   X   X   X   X   X   X       0       X       X  X
> X   X   X   0   X   X   X   X   X   X       1       X       X  X
> X   0   X   X   0   X   X   X   X   X       1       X       X  X
> 1   X   0   X   0   X   X   X   X   X       1       X       X  X
> `#`
> X   X   X   X   X   1   1   X   X   X       X       0       X  X
> X   X   X   1   1   1   X   X   X   X       X       0       X  X
> X   1   1   1   X   1   X   X   X   X       X       0       X  X
> 0   1   X   1   X   1   X   X   X   X       X       0       X  X
> X   X   X   X   X   0   X   X   X   X       X       1       X  X
> X   X   X   0   X   X   0   X   X   X       X       1       X  X
> X   0   X   X   0   X   0   X   X   X       X       1       X  X
> 1   X   0   X   0   X   0   X   X   X       X       1       X  X
> `#`
> X   X   X   X   X   X   X   1   1   X       X       X       1  X
> X   X   X   X   X   1   1   1   X   X       X       X       1  X
> X   X   X   1   1   1   X   1   X   X       X       X       1  X
> X   1   X   1   X   1   X   1   X   X       X       X       1  X
> X   X   X   X   X   X   X   0   X   X       X       X       0  X
> X   X   X   X   X   0   X   X   0   X       X       X       0  X
> X   X   X   0   X   X   0   X   0   X       X       X       0  X
> X   0   X   X   0   X   0   X   0   X       X       X       0  X
> `#`
> X   X   1   X   X   X   X   X   X   X       X       X       X  1
> X   X   X   X   1   X   X   X   X   X       X       X       X  1
> X   X   X   X   X   X   1   X   X   X       X       X       X  1
> X   X   X   X   X   X   X   X   1   X       X       X       X  1
> X   X   0   X   0   X   0   X   0   X       X       X       X  0
>

chriscamacho, 2018-09-14:
just a thought would it be just as valid a test (ie equivalent) if all the *INPUT* X's were replaced with 0's ?

hneemann, 2018-09-14:
> ok it passes this test but says too many entries, does that mean its skipped some tests??

No, no tests are skipped. But there are so many test results that not all results are stored, to avoid a out of memory error. But all tests are executed, and all tests are passed.

> just a thought would it be just as valid a test (ie equivalent) if all the INPUT X's were replaced with 0's ?

That's not the case! Don't replace the X's  by 0's!

If X's at input signals are used in a line in the truth table, this line is expanded to all possible combination of values which are inserted at the X's.

A single line

```
A B C D   Y 
1 X X X   1  
```
is expanded to 8 test cases:
```
A B C D   Y 
1 0 0 0   1  
1 0 0 1   1  
1 0 1 0   1  
1 0 1 1   1  
1 1 0 0   1  
1 1 0 1   1  
1 1 1 0   1  
1 1 1 1   1  
```
If `n` input signals are marked with a X, there are `2^n` test cases generated.

> had to remove the GND and VCC pins connected to a power component before it would run the test, not sure what I'm doing wrong...

Insert the GND and VCC pins using "Edit -> Special 74xx Functions -> Add power supply".

hneemann, 2018-09-15:
I have added a integration [test](https://github.com/hneemann/Digital/tree/master/src/test/resources/dig/test/74xx) for 74181 and 74182. 
I'm pretty sure the 74182 is working properly.

closed 2018-09-15 by hneemann

chriscamacho, 2018-09-15:
great! :) while of course I will send you any chips I end up needing for my own design and testing, you can always contact me directly with a wish list, not sure when I'll fit them in but turning a logic diagram into a circuit can be quite relaxing and diverting while the wife is watching soaps...

commit f4507dad (2018-09-15): adds the classic 74182, closes #188