hneemann/Digital #189: add export to c code

hneemann, 2018-09-15:
Allowing to export a circuit to c code.
Allows to use a Digital circuit in a c program running on a micro controller.
See discussion here: #186

hneemann, 2018-11-01:
I thought about it and decided not to include such a feature in Digital.
In order to operate a circuit that not only contains combinatorial logic, the simulation algorithm, which is also implemented in Digital, must be re-implemented in C. This is a relatively high effort, which has only a very small benefit.

closed 2018-11-01 by hneemann

mgriebling, 2018-11-02:
If you could give some pointers, I'll implement it.

hneemann, 2018-11-02:
I would try to implement a new HDL similar to VHDL or Verilog. The template engine can be used to create most of the code. Each components implementation requires a `readInputs` and `writeOutputs` method. This is necessary to simulate a circuit in which all components react in parallel at the same time. A simulation step is calculated by calling the `readInputs` method for all components where an input has changed. Then all `writeOutputs` methods must be called. During the call of the `writeOutputs` methods, new components will observe a change in their inputs. These can then be used to calculate the next step until there are no more changes. Then the circuit has stabilized.
This procedure should be repeated whenever an input changes.