hneemann/Digital #190: Mac OSX top menus are non-standard

mgriebling, 2018-09-17:
I'm running Digital under OSX 10.13.6 and am seeing the Digital menu at the top of the window (see below).  They should appear where the "Main" is shown.

[image]

Would it be possible to have these menus rendered in the top of the screen like all other Mac Apps.  For example the Finder menus show as follows:

[image]

nevercast, 2018-09-17:
I believe the property required for that is
`System.setProperty("apple.laf.useScreenMenuBar", "true");`
and it needs to be set before the UI is put together.

hneemann, 2018-09-17:
@mgriebling 
I don't know much about MacOS but you can try the option @nevercast has suggested:
Open the bash in the folder which contains the Digital.jar file and try:

```
java -Dapple.laf.useScreenMenuBar=true -jar Digital.jar
```

mgriebling, 2018-09-17:
Neither of these suggestions seems to be working.

nevercast, 2018-09-18:
@hneemann I presume this is a JMenuBar?

mgriebling, 2018-09-18:
I just tried Digital on an older OSX 10.11.6 and see the same thing.  As well, Digital works when invoked via the GUI but I can't seem to run it from the command line under OSX 10.11.6.  The error is "Unsupported major.minor version 52.0.  Java version is openjdk version 1.7.0-ea.  I'll try a few more things under OSX 10.13.x to see if I can fix the issue.

hneemann, 2018-09-18:
@mgriebling 

> The error is "Unsupported major.minor version 52.0". Java version is openjdk version 1.7.0-ea.

Digital needs at least Java 1.8. It does not run with Java 1.7.

mgriebling, 2018-09-18:
Here's some code that appears to do the job of showing the menus properly added to the main() static function:

 ```
       String lcOSName = System.getProperty("os.name").toLowerCase();
        boolean IS_MAC = lcOSName.startsWith("mac os x");
        if (IS_MAC) {
            System.setProperty("apple.laf.useScreenMenuBar", "true");
            System.setProperty("com.apple.mrj.application.apple.menu.about.name", "Digital");
        }
        
        try { // enforce MetalLookAndFeel
            //UIManager.setLookAndFeel("javax.swing.plaf.metal.MetalLookAndFeel");
            UIManager.setLookAndFeel(UIManager.getSystemLookAndFeelClassName());
        } catch (ClassNotFoundException | InstantiationException | UnsupportedLookAndFeelException | IllegalAccessException e) {
            e.printStackTrace();
        }
```

Unfortunately, changing the name via the  `System.setProperty("com.apple.mrj.application.apple.menu.about.name", "Digital");` does not seem to work -- I still get Main as the name.  I'll investigate how to get this work some more.  I believe the key to getting the menus to move to the top may be to use the native look and feel.

Here's how it looks now:

[image]

Note: This is with the "Use dark menus" enabled in preferences.  The shapes look kind of funny in this mode -- stick figure-like.  The default "white menus" look fine.

hneemann, 2018-09-18:
@mgriebling 
If you are using the SystemLookAndFeel please also experiment with the component tree view.
In the past i had problems with the behavior of this component on MacOS.

nevercast, 2018-09-18:
JInternalFrame can have a different look and feel to the root JFrame, wrapping the content in an InternalFrame may be an option to get both the OSX Menu, and still use the Swing look&feel for the component tree view. 

Presuming that the component tree does not render correctly in native.

mgriebling, 2018-09-19:
I see what you mean about the tree view:

[image]

The menu view of the list looks fine.

[image]

mgriebling, 2018-09-19:
Here's how LogiSim appears to work properly with menus and tree view:

[image]

Perhaps we can see what they do?  But it does look like they are limiting the symbol sizes to the text size in the tree view.  That may be the problem in the Mac tree view --- it appears to be only using the text to size the tree view items.

hneemann, 2018-09-19:
@mgriebling 
Looking at your screenshot which shows the menu and the tooltip, it seems to me that also html rendering is not supported by the native apple look and feel implementation.

mgriebling, 2018-09-19:
Yes i noticed that as well.

hneemann, 2018-09-20:
I tend not to support native MacOS look and feel.

- There are differences in the behavior of different look and feel implementations which needs to be worked around. Because there is a lot of code, this requires also a lot of work.
- A lot of testing is necessary to ensure everything works like expected. And testing is difficult if the necessary hardware is not available. This kind of GUI testing is also hard to automate.
- Even if someone else does the implementation and testing, the future development of GUI components is more complicated if multiple look-and-feel implementations need to be supported.

mgriebling, 2018-09-27:
I managed to fix the problem with the JTree view with the following code:

```
 public SelectTree(LibraryTreeModel model, CircuitComponent component, ShapeFactory shapeFactory, InsertHistory insertHistory) {
        super(model);
        this.setRowHeight(0);
        this.shapeFactory = shapeFactory;
        setSelectionModel(null);
```

The key line is to set the row height to zero (0).

Here's how the tree looks now using Mac native mode:

[image]

hneemann, 2018-09-28:
That's interesting! 
This also works for the MetalLookAndFeel, which allows me to add this modification to the main branch.
Could you also figure out how HTML rendering is enabled?

mgriebling, 2018-09-29:
> Could you also figure out how HTML rendering is enabled?

Yes, I'll be having a look at this next.

Atcold, 2025-04-13:
@mgriebling, is your fork fixing the MacOS menu ready to merge?
It looks great!