hneemann/Digital #192: Errors with multiple circuits with the same name

mgriebling, 2018-10-01:
While editing several libraries, I keep getting the following annoying error message:

[image]

Would it be possible for Digital to just replace a previous version of the same name with the newly opened circuit instead of posting this message every time?  This gets very annoying after the first 100 times I see it.  😣

mgriebling, 2018-10-01:
I also encountered the situation where, because of multiple name conflicts, no symbol at all was populated in a circuit that required it.  To fix this issue, I had to open a new instance of Digital and then the sub-circuit was populated correctly.  This seems like a bug.  

BTW, I have not added any circuits, these are all the existing circuits that Digital seems to accumulate as I was fixing the "wideGate" flag.  I would open circuits in the test directories and then the src directories.  Digital seems to be accumulating the paths as I opened new circuits with the same name.

mgriebling, 2018-10-02:
Maybe I messed with the library path.  I'll double-check.

mgriebling, 2018-10-02:
Ok, now I'm sure something was set incorrectly.  I now get multiple errors with duplicate components during the initial build:

```
testProcessorInSimulator(de.neemann.digital.hdl.verilog2.VerilogSimulatorTest)  Time elapsed: 0.006 sec  <<< ERROR!
de.neemann.digital.draw.library.ElementNotFoundException: Error importing the circuit GPO.dig!
	at de.neemann.digital.hdl.verilog2.VerilogSimulatorTest.checkVerilogExport(VerilogSimulatorTest.java:117)
	at de.neemann.digital.hdl.verilog2.VerilogSimulatorTest.testProcessorInSimulator(VerilogSimulatorTest.java:79)
Caused by: java.io.IOException: The file GPO.dig exists multiple times below /Users/michaelgriebling/Documents/Course/Digital/src/test/resources/../../main/dig/processor.
	at de.neemann.digital.hdl.verilog2.VerilogSimulatorTest.checkVerilogExport(VerilogSimulatorTest.java:117)
	at de.neemann.digital.hdl.verilog2.VerilogSimulatorTest.testProcessorInSimulator(VerilogSimulatorTest.java:79)

```

hneemann, 2018-10-02:
And does the file GPO.dig exist multiple times under **/Users/michaelgriebling/Documents/Course/Digital/src/test/resources/../../main/dig/processor**?

mgriebling, 2018-10-03:
As far as I can tell it only appears once.

hneemann, 2018-10-03:
If the folder /Users/michaelgriebling/Documents/Course/Digital/src/test/resources/../../main/dig/processor contains only one copy of GPO.dig that's a bug.
Which folder is used as library folder in the settings?

mgriebling, 2018-10-03:
I've been able to replicate this on a second computer with a fresh Digital clone.  It appears that if the Library is set to something like  **/mikeg/Documents/Course/Digital/src/main/dig** then I get all sorts of errors during the build.

When I change it back to the default directory **/Users/mikeg/Documents/Course/Digital/src/main/dig/lib** everything is fine.  I also don't get the duplicate parts pop-up.

hneemann, 2018-10-03:
That's what I would expect.
The folder main/dig contains all kinds of duplicate file names. This folder is not suited to be used as a library which requires all file names to be unique.

mgriebling, 2018-10-03:
It's perhaps not obvious to a newbie that changing the directory could cause such issues. 😖 A brief mention in the user manual might warn of the dangers of changing the Library directory.