hneemann/Digital #193: Maven build error with generated files

mgriebling, 2018-10-01:
I'm getting the following strange maven errors when it attempts to move generated files:

_[WARNING] Assembly file: /Users/michaelgriebling/Documents/Course/Digital/target/Digital.jar is not a regular file (it may be a directory). It cannot be attached to the project build for installation or deployment._

Yet, this is a file that maven generated.  Anyone know if there is a way to fix this?  I'm using the latest maven release 3.5.4 under OSX 10.14.  This is a cloned project directly from this site.

hneemann, 2018-10-02:
It's a warning, not an error. And this states that the generated file is neither suitable for installation in the local maven repository nor for deployment in the maven central repository.
And that's fine, because the file is not intended to be stored there.

mgriebling, 2018-10-02:
Good to know.  I was wondering why the distribution directory doesn't get populated.  I assume this needs to be manually?

hneemann, 2018-10-02:
distribution directory?

mgriebling, 2018-10-02:
Yes, the Digital/distribution directory that contains the Digital.exe, checklist.txt, etc.  In normal Java releases this directory is filled with the .java and other files to be distributed.  Maybe maven works differently.

hneemann, 2018-10-02:
I don't know how a "normal" Java project looks like. ;-)
But the maven project structure is very common in the Java world.

mgriebling, 2018-10-03:
Here's a typical Java project (default project built by Netbeans):

[image]

I'm new to Maven so learning as I go. 😉  It seems that most Java people are going with JavaFX, which is why it's difficult to find Mac OSX examples using Swing.  I did find an interface called VAqua, but unfortunately it's not in the Maven universe so I had to fudge things with a manual Maven dependency install.  But the GUI looks much better under OSX.  It even supports the most recent Mac "dark mode".

closed 2018-10-04 by hneemann