hneemann/Digital #195: Error while simulating circuit

mgriebling, 2018-10-04:
I'm in the process of transferring some of my Logisim circuits to Digital and encountered the following error during simulation:

[image]

Here's the file that caused the problem.  Set all the inputs high, enable the alarm, and set one of the inputs low to reproduce the error.  I don't believe there are multiple outputs feeding one input.

[NandAlarm.dig.zip](https://github.com/hneemann/Digital/files/2445895/NandAlarm.dig.zip)

hneemann, 2018-10-04:
Try to remove the NAnd which is directly connected to the output.

mgriebling, 2018-10-04:
That's nasty, how did you find that?  I wonder if there is a way to identify where the problem is located to help in debugging this sort of issue.  It certainly isn't visually obvious.

closed 2018-10-04 by mgriebling