hneemann/Digital #205: Too numb for Multiplexers

stehgold, 2018-10-26:
Hi,
I'm trying to build a 4-bit shift register from d-flipflops and multiplexers. Select bits 00 for right shift, 01 for erase, 10 for load and 11 for left shift.
I wired all this up and when I try to simulate I get an error (in German) saying something like:

4 bits needed but 1 bits found. Affected are: commonBusOut, Multiplexer.

[image]
I do not understand what's wrong.

hneemann, 2018-10-26:
Only with the screenshot that's hard to say.
However, flip flops and muxers are able to store or switch more than one bit.
It seems to me that the muxers are configured to switch 4 bit, and the flip flops are configured to store only one bit.
Please zip the file and attach it to this issue. Then I can tell you more.

By the way: the inputs Q0-Q3 should be outputs

hneemann, 2018-10-26:
A wire needs to end or start at a components pin. If a wire is drawn across a pin, the pin is not connected to that wire.

stehgold, 2018-10-26:
Messy wiring but worky :)
[image]

stehgold, 2018-10-27:
It warned me I have connected different outputs and refused to simulate. Maybe some wiring issue. Anyways, my fault was I thought the number of bits is the number of inputs. But that's determined by the number of sel-bits.
Thanks

stehgold, 2018-11-01:
Hi,
new problem here. How does splitter work ? The exercise is to build a 6-bit input 1s counter from a loadable shift-register and a counter. And I need a "Ready"-Flag when all bits are shifted. So, NORing all bits. The shifting works but when I add the NOR I get a "Bit widths don't match" error. I played around with the splitter settings for an hour now but don't get it.

I could get rid of that but got a short circuit when simulating.

I want to split a 6-bit bus into its single bits. What splitter settings do I need ?

The last I tried was:
Input: 1*6
Output 1*6

Stephan

hneemann, 2018-11-01:
Input: `6`
Output: `1*6`

or

Input: `6`
Output: `1,1,1,1,1,1`

Then you get:

[image]