hneemann/Digital #207: Is there a component representation case or casex?

itviewer, 2018-10-30:
Is there a condition matching structure (case or casex)?

hneemann, 2018-10-30:
The component that comes closest to the verilog case statement is the multiplexer.

itviewer, 2018-10-31:
It seems that only the ‘test case’ component can input x(don't care),
No any IO component can enter x as a multiplexer option？

hneemann, 2018-10-31:
You have to imagine the simulation as a real circuit and the wires as real wires.
A real wire can be connected to 0 volts (0,low) or 5 volts (1,high). The wire can also remain unconnected (high z). But a real wire can not be connected to the voltage "don't care".
But the circuit to which the wire is connected may be independent of the state of the wire.

itviewer, 2018-11-01:
Thanks, I want to present such a circuit structure (support z or don't care )
```
    casez(sel)
        3'b001:    y=a;
        3'b010:    y=b;
        3'b1??:    y=c;
        default:  y=a;
    endcase
```
there seems to be no component that can be used directly, and finally I use external verilog component.

closed 2018-11-01 by itviewer

hneemann, 2018-11-01:
This function is easy to implement with a simple muxer:

[image]

Although it is possible to use components defined in verilog or vhdl, this should be avoided if it is not required for didactic reasons. The simulation of a circuit containing vhdl or verilog defined components is much slower compared to a circuit containing only normal components.