hneemann/Digital #214: Registers without reset should not be initialized to zero

fdedinec, 2018-11-15:
I love that Digital gets right most of what was wrong in Logisim in terms of pedagogy (dangling inputs trigger an error, etc).
After a while I finally found one feature that I disagree with: a register (flip flop or whatever) without reset pin is initialized with zero value. For me (as a teacher) it is almost as bad as a dangling input. 
For instance, a counter where you forget the reset should not work. The value should be unknown ('U' in VHDL) with a corresponding non-green color.
I see that there is a branch that considers six-valued logic, maybe it will be fixed there.
In any case thanks for the excellent work.

hneemann, 2018-11-15:
That's an interesting idea. But a value "unknown" makes no sense to me. In a real circuit the value is always defined (0 or 1). It may be that it can not be calculated, but it is always either 0 or 1.
In this case, the value should be chosen randomly at startup.
But such a flip-flop can already be easily constructed. Because when a flip-flop is built from individual gates, the initial state is actually random. A reset input is required to bring it to a defined state.

fdedinec, 2018-11-15:
Thanks for the immediate feedback
> In this case, the value should be chosen randomly at startup.

I'd rather a big flashing error message but this would be a very satisfactory solution, and quick to implement.
Another motivation: My next lecture is about automata. An FSM must include an initial state, otherwise it is under-defined. And the initial state will not always be encoded as "all zeroes" -- for instance in one-hot encoding, one of the state bits must be set (and only one).
Experience show that students tend to forget to specify initial state, and I want things to go wrong in this case. 
Your solution of random initial state has some elegance in that it will go horribly wrong (sometimes it works and it is difficult to reproduce), indeed as would a real circuit.
(then they will more likely blame a Digital bug than a bug in their design but this is another story)

hneemann, 2018-11-15:
Do you still believe that something should be changed in the built-in components?
Or do you consider a solution like [this](https://github.com/hneemann/Digital/files/2586643/ff.zip) sufficient?

closed 2018-11-22 by hneemann