hneemann/Digital #215: floating unused pins

mitolegy, 2018-11-20:
if theres allready a solution for this im sorry but ive noticed that most components will not run with some pins floating. logisim solves this nicely by allowing pins to pull up or down by themselvs but digital just asks you to connect it to something.
this may not sound like a problem but its really irratating when i create a circuit im going to copy and paste all over the place and i have to ether delete the model im copying from if i want to test the circuit or spend a lot of time removing pull up and pull down resistors. idk if it would be a pain to add. its just that this is one of the main things that are making the switch from logisim to digital hard.

hneemann, 2018-11-20:
See also the discussion in #36.
It is bad practice not to connect unused pins. Therefore, the simulator does not allow this.
However, you do not need to add pull-down or pull-up resistors. You can simply use the VCC or GND components to connect an input to high or low.

[image]

hneemann, 2018-11-21:
I have already thought about it a lot: I do not want to allow circuits with open inputs.
Especially when used in teaching, allowing open inputs makes no sense to me.
But you're right: this makes the creation of a circuit a bit more cumbersome.
For me that's less important in this case.

closed 2018-11-22 by hneemann

Johnlon, 2019-07-22:
Personally I think *requiring* the student to select a default for unused pins if they want to elide those, either at the device or circuit level, would be entirely sufficient to remind everyone that this is an important detail. My suggestion seems sufficient for the circuit to have an unambiguous defined value for all inputs without excessive overhead and without sacrificing the principal.

hneemann, 2019-07-23:
@Johnlon 
Do you more often use the 74xx circuits or the built-in components?