hneemann/Digital #219: Adjust position of the label text according to the rotation and direction of the port when import svg file

itviewer, 2018-12-02:
Hi,
Thank you very much for the efforts in importing SVG files.
Instead of just depending on the direction(input、output) that the text is allowed only to the left or right of the port shape, If it can automatically adjust the appropriate label position according to the rotation and direction of the port, which will allowing four-sided components, that will be perfect.

hneemann, 2018-12-02:
It is possible to turn off the automatic labeling (remove the "+" from the pin circle id).
Then you can add the labels in the SVG file as you wish.

itviewer, 2018-12-03:
Ok, By manually add the port's label,  we have to adjust the size and color of the label text to keep the same style as the default. It seems that we can only do it in this way now. 
Anyway, importing SVG files is very powerful and flexible, Thanks.

hneemann, 2018-12-03:
Hmm! Maybe I should add the labels to the template. 
Then it is easy to keep them if you wish.

itviewer, 2018-12-03:
> Hmm! Maybe I should add the labels to the template.
> Then it is easy to keep them if you wish.

Yes!  it will be very convenient that we only need to rotate  the label text as needed.