hneemann/Digital #225: [feature request] raw binary loading to RAM

Houkime, 2018-12-16:
"Hex files" that are outputted by assembler IDE referenced in README might be great, however that's not what you always want when tinkering with alternative ISAs.

Basically what I need right now for my mini RSSB OISC cpu is an ability to load an arbitrary binary file into RAM.
So that I can use an assembler which spews out an execution-ready code in its standard form - binary and later have a usual C toolchain based atop of that.

I think that the easiest way to do that since Digital consumes "hex files" would be to use sth like an os-provided hexdump utility (which is called just "hexdump" on any *NIX/BSD system I've used).

Right now because of the exclusive usage of "hex file" format I probably need to write a custom script to convert my binaries to it which is suboptimal and I see it is an unneccessary extra step for a user.

Houkime, 2018-12-17:
It seems like because of exclusive Digital-Assembler relationships and not fully documented format specification (which is not Intel-hex despite misleadingly being named .hex) ram loading for bit-numbers other than 16 is not really supported which is a large bummer given that a circuit logic sim shouldn't assume anything about bitnumbers beforehand cause circuits are freeform.

hneemann, 2018-12-17:
> Basically what I need right now for my mini RSSB OISC cpu is an ability to load an arbitrary binary file into RAM.

The hexdump tool you mentioned can be used to convert an arbitrary binary file to a hex format which can be imported by Digital:

To convert "z.bin" to "z.hex" expecting a 8 bit memory layout:

`
echo "v2.0 raw" >z.hex && hexdump -v -e '"" 1/1 "%x" "\n"' z.bin >>z.hex
`

> It seems like because of exclusive Digital-Assembler relationships and not fully documented format specification (which is not Intel-hex despite misleadingly being named .hex) ram loading for [...]

A digital hex file is very simple: A single hex value per line is associated with a single memory location. The bit width of the memory does not matter. A 8-bit memory works just like a 16-bit or 64-bit memory does.

Houkime, 2018-12-17:
Thanks.
To avoid further confusion, could you add a link to .hex specification and this hexdump line to the ram help/ram loading menu so that it is evident?

commit eac5ea91 (2018-12-17): Added BIN and Intel HEX support for the RAM/ROM content loader. Closes #225
closed 2018-12-17 by hneemann