hneemann/Digital #227: GUI Error (Overdraw?) on Windows

JazzaHat, 2019-01-09:
Hi,
I've had this issue for a while now: the GUI seems to overdraw, I believe the right word is. Sometimes this doesn't show up for minutes, sometimes its immediate. Dragging the window to a different screen "fixes" the GUI until it errors again. The program itself seems to continue working and saving just fine.
[image]
I also have this issue in Logisim.
I have tried multiple Digital versions, uninstalling and reinstalling of: Java, Graphics drivers.
I have also recently upgraded my GPU and have a fresh install of windows.
I am not sure if there are any alternative settings I can use, but I have tried looking in the settings menu and I'm unable to find anything that seems to help.
This might not be a problem with Digital, however it (and Logisim) are the only two programs that reproduce this issue for me. Any ideas?

hneemann, 2019-01-10:
I have added a new EXE wrapper which turns of D3D. Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).
There is a new EXE wrapper called "Digital_noD3D.exe".
Please remove your J2D_D3D = false environment variable and try the new Wrapper.
If every thing works like expected, using the Digital.exe wrapper should still show the problem and using the Digital_noD3D.exe wrapper should fix it.

JazzaHat, 2019-01-10:
Yup, Digital.exe does show the problem and Digital_noD3D.exe didn't appear to show the problem :) 
Thank you very much!

JazzaHat, 2019-01-11:
Actually it still appears to be a thing. I was using it for the past few hours fine, then I opened up a second window. I am definitely using the noD3D version.

[image]

I guess the current work around is to only have one instance open at once?

If there's any other debug information you want let me know.

Thanks, Jazza

hneemann, 2019-01-15:
To be honest: I have no idea what is going on on your machine!
Did you turn on display scaling? Maybe it is this [bug](https://bugs.openjdk.java.net/browse/JDK-8187367).

inacioalves, 2020-02-22:
[image]
Hi guys, I just start to trying the Digital and I found this problem. The logic ports icons don't have correct appearance.
There is any way to solve this?

Thanks in advance.

hneemann, 2020-02-22:
I the settings you can select IEEE 91-1984 shapes.

Roger13h, 2022-12-24:
IM hitting this issue ,, after 5 min or so of use ..  display elements seem to be off on the display ,

THplusplus, 2025-03-28:
Unfortunately, I'm still experiencing graphical glitches in the UI:

[image]

This is easily reproducible:

1. start Digital.exe (or Digital_noD3D.exe, problem occurs with either)
2. add a simple "RAM, Chip Select" component
3. right click it to open its properties
4. Select any value from the "Data Bits" drop down
5. No the UI is glitched. Hovering over different UI elements with the mouse will immediately cause graphical glitches as shown in the screenshot

Environment:
* Windows 11 Pro 24H2 (26100.3476)
* Java Version 8, Update 441

PS: I was just about to send this post like this, then I got it working by starting it with

`java -Dsun.java2d.d3d=false -jar Digital.jar` <= THIS WORKS for me

I had tried

`java -jar Digital.jar -Dsun.java2d.d3d=false` before, which didn't work. I'm no Java guy, I don't know that the order of the command line arguments matters here.

Can't you make sun.java2d.d3d=false the default internally? I was SO close to giving up on what I'm sure is actually a great tool, I really only got saved by luck here. I bet this has driven way A LOT of first time users.